JSubelj/g910-gkey-macro-support #1: AUR package?

ShayBox, 2018-12-16:
Could you make an AUR package that builds this on arch.

Really cool project, thanks for supporting this keyboard.

frostworx, 2018-12-25:
after having used my slighly modified local version of
https://github.com/CReimer/g910-gkey-uinput
I decided today to switch to this project (not just because the script didn't work today ;)).
thanks to the author!
just rolled my own worksforme PKGBUILD:

`pkgname=g910-gkey-macro-support
pkgver=0.1.0
pkgrel=1
pkgdesc='GKey support for Logitech G910 Keyboard on Linux'
url='https://github.com/JSubelj/g910-gkey-macro-support'
arch=('any')
license=('MIT')
depends=('python-pyusb' 'python-uinput')
source=('git://github.com/JSubelj/g910-gkey-macro-support.git')

package() {
  cd ${srcdir}/${pkgname}
  python setup.py install --root="${pkgdir}"
  install -Dm 644 -t $pkgdir/usr/lib/systemd/system g910-gkeys.service
}
md5sums=('SKIP')`

frostworx, 2018-12-25:
hm, I'm afraid I botched using the "Insert Code" button :)

JSubelj, 2019-01-03:
Thank you @frostworx for that, will definitely try it out. Also did you put it on aur, because I don't seem to find it.

frostworx, 2019-01-03:
nah, I haven't put it on aur yet. I'm more or less new on arch (after ~14 years gentoo) and don't know very much about pkgbuilds yet :) your tools works still very fine and I'm glad you made it :)

JSubelj, 2019-01-04:
No problem I will probably put it on aur when I have a bit more time and your PKGBUILD will seriously help me a lot.

JSubelj, 2019-02-25:
I added the PKGBUILD. It's located in pkgbuild dir. Thank you once again @frostworx your initial PKGBUILD really helped. I will probably push it to AUR when I see the driver is (mostly) bug free. Try it out. If you have any problems you are welcome to write about it here (same topic) or open a new issue request. Have fun.

closed 2019-02-25 by JSubelj