JSubelj/g910-gkey-macro-support #5: Ubuntu 18.04 - installer.sh issues

braoult, 2018-12-28:
I installed g910-gkey-macro-support via installer.sh (python 2.7). However there is a mismatch between the install path (g910-gkeys is installed in /usr/local/bin) and the systemd setup, which tries to launch the (non-existent) /usr/bin/g910-gkeys.

However, even after changing path in /etc/systemd/system/g910-gkeys.service, there is a python error (likely a missing dependency) when I try to start the service (unfortunately I do not know anything about python):
```
$ sudo systemctl start g910-gkeys
$ ps auxww | grep  g910 | grep -v grep
$ journalctl -u g910-gkeys
[...]
Dec 28 11:43:36 lorien systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Dec 28 11:43:36 lorien g910-gkeys[3123]: Traceback (most recent call last):
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "/usr/local/bin/g910-gkeys", line 11, in <module>
Dec 28 11:43:36 lorien g910-gkeys[3123]:     load_entry_point('g910-gkey-macro-support==0.1.0', 'console_scripts', 'g910-gkeys')()
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
Dec 28 11:43:36 lorien g910-gkeys[3123]:     return get_distribution(dist).load_entry_point(group, name)
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
Dec 28 11:43:36 lorien g910-gkeys[3123]:     return ep.load()
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2324, in load
Dec 28 11:43:36 lorien g910-gkeys[3123]:     return self.resolve()
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
Dec 28 11:43:36 lorien g910-gkeys[3123]:     module = __import__(self.module_name, fromlist=['__name__'], level=0)
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "build/bdist.linux-x86_64/egg/cli_entry_point.py", line 4, in <module>
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "build/bdist.linux-x86_64/egg/lib/misc/logger.py", line 2, in <module>
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "build/bdist.linux-x86_64/egg/lib/misc/paths.py", line 4, in <module>
Dec 28 11:43:36 lorien g910-gkeys[3123]:   File "build/bdist.linux-x86_64/egg/lib/misc/create_config.py", line 6, in <module>
Dec 28 11:43:36 lorien g910-gkeys[3123]: ImportError: cannot import name paths
Dec 28 11:43:36 lorien systemd[1]: g910-gkeys.service: Main process exited, code=exited, status=1/FAILURE
Dec 28 11:43:36 lorien systemd[1]: g910-gkeys.service: Failed with result 'exit-code'.
```

JSubelj, 2019-01-03:
Hey,
first of all thank you for trying to use my driver.
Now for the issue. It is clearly stated that you need to use python3.7 (the code is also written with this version in mind and uses loads of stuff that python2.x does not have). So please try to install it with python 3.7 and then if you still don't get the expected results, I will try to help you to my best extend.

Also sorry for late response (I thought Github sends you an email when you get an issue posted).

Have fun

JSubelj, 2019-01-05:
Very happy to hear that. The path still points to /usr/bin/, because the g910-gkeys.service is just copy pasted from the repo to /etc/systemd/system/. I will probably change that so that the .service file is generated after install. Thank you for submitting this issue and giving me ideas on how to improve my software.

Mace68, 2019-06-17:
I had similar issues on Ubuntu 18.04 but have it working now:

line 5 of installer.sh
`cp g910-gkeys.service /usr/lib/systemd/system`
actually creates a file called system in /usr/lib/systemd/
I removed that file and manually copied g910-gkeys.service to /etc/systemd/system/

I also, as did braoult, had to change ExecStart in /etc/systemd/system/g910-gkeys.service from /usr/bin/g910-gkeys to /usr/local/bin/g910-gkeys

Thank you very much for your work on getting the G910 GKeys working :)

wraith3690, 2019-10-17:
thank you JSubelj for making this XD and I can confirm on mint 19.2 that the changes mace68 has done get it working