JSubelj/g910-gkey-macro-support #6: Allow G-keys to be considered as a modifier.

braoult, 2019-01-05:
Hi,

I made a test (Ubuntu 18.04, python 3.6) and everything looks fine for now...

A nice improvement (if possible) would be to allow g-keys to be considered as modifiers (ctrl, meta, super, hyper) or even a combination of some modifiers (such as ctrl+super).
For example, by having in configuration file something like:
```
    "g4": {
        "hotkey_type": "modifier",
        "do": "ctrl"
    },
    "g5": {
        "hotkey_type": "modifier",
        "shortcut": "ctrl+super"
    },
```
I would get ctrl-c when typing G4+c, or ctrl-super-t when typing G5+t

I am not sure it is feasible, but I am pretty sure emacs users would find some usage and save their fingers ;-)

Thanks again for this wonderful tool !

JSubelj, 2019-01-05:
Currently there is no support for that, because the driver is not catching GKey release signal, but I will probably add it after the uni exams. Thank you for your suggestion.

braoult, 2019-01-08:
Thanks Jan (is it your firstname? sorry if I made a mistake),

Oh yes, I noticed there is no key release event (I made tests adding F13-F21 in char_uinput_mapper.py, mapping the G-keys to them, and expecting to manage everything at X11 or app level, but, with no "real" release event (xev shows both KeyPress and KeyRelease are generated at same time on keypress action), there are some limitations, including modifiers and repeat.

JSubelj, 2019-01-08:
Yeah it's Jan :). Glad to hear that you made it happen with the Fkeys. If you like you can also create a pull request and I'll test it out and merge it. 

About the release event. In the past I was also thinking about putting release events in so in the next release I'm almost certain I'll put it in. Probably the release events will be emitted only on `shortcut` hotkey type (because having them on `typeout` or `run` doesn't really make sense). 

The only problem I see is that you could still use just one GKey at a time. This is due to keyboard sending other hex codes when two GKeys are pressed together (you can see that if you tail /var/log/g910-gkeys.log and press one GKey while holding down another). In the future I will investigate that further if you want, maybe find out a pattern and reverse engineer it so that you can bind ex. ctrl to G1 and alt to G2 and press them down together.

For G8/F20 please show code in char_uinput_mapper.py, config.json and log output when you press it and maybe I will be able to help you.

JSubelj, 2019-02-25:
I just added using Gkeys as modifiers. Try it out. If you have any problems you are welcome to write about it here (same topic) or open a new issue request. Have fun.

closed 2019-02-25 by JSubelj