JSubelj/g910-gkey-macro-support #8: [Question] Numpad Keys?

gviscardi, 2019-02-14:
Does this excellent software support numpad keys? Like if I wanted alt+NumPad5, would that work?

JSubelj, 2019-02-25:
Hey, sorry for the late response. First of all thank you for using my software and taking the time to ask a question. 

Currently numpad keys are not supported but you can easily add this functionality by adding them in `char_uinput_mapper.py` (and also make a pull request so I can add it as well :D).

Currently the buttons are automatically released when you click a GKey (I implemented it that way because of simplicity but because you're not the only one https://github.com/JSubelj/g910-gkey-macro-support/issues/6 I will most likely fix that).

gviscardi, 2019-02-25:
I will see what I can do about a pull request with the numpad key support :)

And the ability for the GKeys to be able to be held down would be very useful for things like PTT on voice applications. Thank you for your support!

JSubelj, 2019-02-25:
I just added numpad keys and using Gkeys as modifiers. Try it out. If you have any problems you are welcome to write about it here (same topic) or open a new issue request. Have fun.

closed 2019-02-25 by JSubelj