JSubelj/g910-gkey-macro-support #10: [Issue]

Gitnotrekt, 2019-03-24:
**Describe the bug**
When I run chmod +x installer.sh; sudo ./installer.sh or sudo python setup.py install --record files.txt it gives me an error code:" Traceback (most recent call last):
  File "setup.py", line 8, in <module>
    import lib.PROJECT_INFO as PROJECT_INFO
  File "/home/leon/Schreibtisch/g910key/g910-gkey-macro-support/lib/PROJECT_INFO.py", line 5
SyntaxError: Non-ASCII character '\xc5' in file /home/leon/Schreibtisch/g910key/g910-gkey-macro-support/lib/PROJECT_INFO.py on line 5, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
"
, and that's why I can't get it to install.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://github.com/JSubelj/g910-gkey-macro-support/wiki and follow the steps.'
2. Click on '....'
3. Scroll down to 'Installation'
4. See error

**Expected behavior**
It should install without any problems.

**Screenshots**
[image]

**Additional info**
 - did you run in daemon or normal mode: It can't even install.
 - logs from /var/log/g910-gkeys.log: Non existent.
 - logs from jurnalctl -u g910-gkeys: Don't know.
 - configuration in /etc/g910-gkeys/config.json: Non existent.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.10
 - 4.20.13 (because older kernels don't work with my graphics card: RX590)
 - gnome
 - 0.2.2
 - 3.7

**Additional context**
When I installed uinput I could only run the python install command with sudo to install it. (http://tjjr.fi/sw/python-uinput/#download-and-install at "Download and install", see image read the error code.)
[image]

JSubelj, 2019-03-25:
Hey, thank you for a thorough report. As i can see from the second screenshot you are using python2.7 and not python3. This can be also tested with `python --version`. Python 3.7 is needed for the driver to run. This was already reported in issue: https://github.com/JSubelj/g910-gkey-macro-support/issues/3 and issue: https://github.com/JSubelj/g910-gkey-macro-support/issues/5
Try the steps that are written in those two issues (running `python3 setup.py install` should suffice). 

About running the setup.py command as normal user:
It is normal that you have to run the command as super user (root) because you are installing software, that means that you are tempering with system files which means that you need root privilege.

If the methods in the referenced issues do not work please do not hesitate to reply to this issue and post the error messages.

Thank you for trying my software.

Gitnotrekt, 2019-03-29:
I partly uninstalled Python 2.7 and tried it with the command "python" and "python3.7", but now it gives me an other error code. I will attach the error codes soon.

Gitnotrekt, 2019-03-31:
Ok, here is the error code: 
[image]

I had the g810-led thing installed before I tried installing this program. I could even press the G5 key and my browser window would reload, so I now tried disabling the Gkey to Fkey mapping and now I obviously can't reload my browser windows anymore with G5. But it still doesn't install.

JSubelj, 2019-03-31:
The error says you dont have setuptools installed. Install it first with `sudo pip3 install setuptools` then try `sudo python3 setup.py install`.

Gitnotrekt, 2019-04-06:
Thanks, that worked.

closed 2019-04-06 by Gitnotrekt