JSubelj/g910-gkey-macro-support #11: [Issue] cannot find -lpython3.7m

germangoergs, 2019-03-29:
Hello guys,

when trying to install this great software, I receive the following error message on OpenSuse Tumbelweed:

```
running install
running bdist_egg
running egg_info
writing g910_gkeys.egg-info/PKG-INFO
writing dependency_links to g910_gkeys.egg-info/dependency_links.txt
writing entry points to g910_gkeys.egg-info/entry_points.txt
writing requirements to g910_gkeys.egg-info/requires.txt
writing top-level names to g910_gkeys.egg-info/top_level.txt
reading manifest file 'g910_gkeys.egg-info/SOURCES.txt'
writing manifest file 'g910_gkeys.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/bdist.linux-x86_64/egg
copying build/lib/cli_entry_point.py -> build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/lib
copying build/lib/lib/PROJECT_INFO.py -> build/bdist.linux-x86_64/egg/lib
copying build/lib/lib/__init__.py -> build/bdist.linux-x86_64/egg/lib
copying build/lib/lib/develop_wo_keyboard.py -> build/bdist.linux-x86_64/egg/lib
copying build/lib/lib/g910_gkey_mapper.py -> build/bdist.linux-x86_64/egg/lib
creating build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/__init__.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/char_uinput_mapper.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/command_bytearray.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/config_reader.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/hotkey_type.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/supported_configs.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
copying build/lib/lib/data_mappers/uinput_all_keys.py -> build/bdist.linux-x86_64/egg/lib/data_mappers
creating build/bdist.linux-x86_64/egg/lib/functionalities
copying build/lib/lib/functionalities/__init__.py -> build/bdist.linux-x86_64/egg/lib/functionalities
copying build/lib/lib/functionalities/gkey_functionality.py -> build/bdist.linux-x86_64/egg/lib/functionalities
copying build/lib/lib/functionalities/media_static_keys_functionality.py -> build/bdist.linux-x86_64/egg/lib/functionalities
creating build/bdist.linux-x86_64/egg/lib/keyboard_initialization
copying build/lib/lib/keyboard_initialization/__init__.py -> build/bdist.linux-x86_64/egg/lib/keyboard_initialization
copying build/lib/lib/keyboard_initialization/usb_and_keyboard_device_init.py -> build/bdist.linux-x86_64/egg/lib/keyboard_initialization
creating build/bdist.linux-x86_64/egg/lib/misc
copying build/lib/lib/misc/__init__.py -> build/bdist.linux-x86_64/egg/lib/misc
copying build/lib/lib/misc/create_config.py -> build/bdist.linux-x86_64/egg/lib/misc
copying build/lib/lib/misc/is_installed.py -> build/bdist.linux-x86_64/egg/lib/misc
copying build/lib/lib/misc/layout_config_helpers.py -> build/bdist.linux-x86_64/egg/lib/misc
copying build/lib/lib/misc/logger.py -> build/bdist.linux-x86_64/egg/lib/misc
copying build/lib/lib/misc/paths.py -> build/bdist.linux-x86_64/egg/lib/misc
creating build/bdist.linux-x86_64/egg/lib/uinput_keyboard
copying build/lib/lib/uinput_keyboard/__init__.py -> build/bdist.linux-x86_64/egg/lib/uinput_keyboard
copying build/lib/lib/uinput_keyboard/keyboard.py -> build/bdist.linux-x86_64/egg/lib/uinput_keyboard
byte-compiling build/bdist.linux-x86_64/egg/cli_entry_point.py to cli_entry_point.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/PROJECT_INFO.py to PROJECT_INFO.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/develop_wo_keyboard.py to develop_wo_keyboard.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/g910_gkey_mapper.py to g910_gkey_mapper.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/char_uinput_mapper.py to char_uinput_mapper.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/command_bytearray.py to command_bytearray.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/config_reader.py to config_reader.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/hotkey_type.py to hotkey_type.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/supported_configs.py to supported_configs.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/data_mappers/uinput_all_keys.py to uinput_all_keys.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/functionalities/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/functionalities/gkey_functionality.py to gkey_functionality.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/functionalities/media_static_keys_functionality.py to media_static_keys_functionality.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/keyboard_initialization/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/keyboard_initialization/usb_and_keyboard_device_init.py to usb_and_keyboard_device_init.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/misc/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/misc/create_config.py to create_config.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/misc/is_installed.py to is_installed.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/misc/layout_config_helpers.py to layout_config_helpers.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/misc/logger.py to logger.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/misc/paths.py to paths.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/uinput_keyboard/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-x86_64/egg/lib/uinput_keyboard/keyboard.py to keyboard.cpython-37.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying g910_gkeys.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying g910_gkeys.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying g910_gkeys.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying g910_gkeys.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying g910_gkeys.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying g910_gkeys.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
lib.functionalities.__pycache__.gkey_functionality.cpython-37: module MAY be using inspect.stack
creating 'dist/g910_gkeys-0.2.2-py3.7.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing g910_gkeys-0.2.2-py3.7.egg
removing '/usr/lib/python3.7/site-packages/g910_gkeys-0.2.2-py3.7.egg' (and everything under it)
creating /usr/lib/python3.7/site-packages/g910_gkeys-0.2.2-py3.7.egg
Extracting g910_gkeys-0.2.2-py3.7.egg to /usr/lib/python3.7/site-packages
g910-gkeys 0.2.2 is already the active version in easy-install.pth
Installing g910-gkeys script to /usr/bin

Installed /usr/lib/python3.7/site-packages/g910_gkeys-0.2.2-py3.7.egg
Processing dependencies for g910-gkeys==0.2.2
Searching for python-uinput==0.11.2
Reading https://pypi.org/simple/python-uinput/
Downloading https://files.pythonhosted.org/packages/54/b7/be7d0e8bbbbd440fef31242974d92d4edd21eb95ed96078b18cf207c7ccb/python-uinput-0.11.2.tar.gz#sha256=99392b676c77b5795b86b7d75274db33fe754fd1e06fb3d58b167c797dc47f0c
Best match: python-uinput 0.11.2
Processing python-uinput-0.11.2.tar.gz
Writing /tmp/easy_install-jfz5_es0/python-uinput-0.11.2/setup.cfg
Running python-uinput-0.11.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-jfz5_es0/python-uinput-0.11.2/egg-dist-tmp-_tb34fsm
/usr/lib64/gcc/x86_64-suse-linux/8/../../../../x86_64-suse-linux/bin/ld: cannot find -lpython3.7m
collect2: error: ld returned 1 exit status
error: Setup script exited with error: command 'gcc' failed with exit status 1
```

I use Python 3.7.2, but I coudn't figure out, what's wrong, hopefully someone else recognizes the problem. :D

Thanks in advance!

Cheers!

JSubelj, 2019-03-30:
Hey, it looks like there are some kind of problems with uinput installing on your system. Have you tried running `pip install python-uinput`? Maybe this will give more extensive error message.

germangoergs, 2019-03-30:
> Hey, it looks like there are some kind of problems with uinput installing on your system. Have you tried running `pip install python-uinput`? Maybe this will give more extensive error message.

Thanks for your fast reply,

you seem to be right, when I try to install it, I get the following error:
`Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-1eocsqie/python-uinput/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-2rx2jhqm/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-1eocsqie/python-uinput/`

I didn't manage to fix it, but maybe you can help me. :)

JSubelj, 2019-03-30:
Don't really have a clue. I guess you did run it as `sudo pip install uinput` as my software did indeed install. 

Maybe try `modprobe uinput` and then pip install. Or you can try installing it with zypper. Found some .rpm files here: http://rpm.pbone.net/index.php3?stat=3&search=python-uinput&srodzaj=3&dist[]=46 

Im sorry if there are grammar mistakes/syntactic mistakes im posting this in the middle of the forrest :P

Also I'm closing this issue because it is not related to my software but to module uinput. But I will try as best as i can to help you install it :)

closed 2019-03-30 by JSubelj

germangoergs, 2019-04-01:
> Don't really have a clue. I guess you did run it as `sudo pip install uinput` as my software did indeed install.
>
> Maybe try `modprobe uinput` and then pip install. Or you can try installing it with [...]

Thanks, got it working; had to install uinput manually, but had to install these packages before:
`sudo zypper install python-devel python-pip gcc`
Afterwards I was able to install uinput.
To install this program here, I looked for "python-devel" on yast and found a package that was called python development or so.
Then I was able to install it correctly. :)