JSubelj/g910-gkey-macro-support #12: [Question] char_uinput_mapper.py limitations

braoult, 2019-04-27:
Hi,

I saw you did implement the modifier keys (I did not test them yet). I have a question, regarding char_uinput_mapper.py. It looks like the accepted keys are those present on en/si keyboards.

My question is: Is there any reason why all /usr/include/linux/input-event-codes.h cannot be used ?
For instance, being able to use KEY_KATAKANA, KEY_MUHENKAN, or any other special key could give another way to use these 9 supplementary keys.

About modifier usage, do you confirm the syntax would be, for instance to have g5 being seen as "KEY_LEFTMETA", so that I could for instance type "G5+c" to do "Meta-c":

```
"g5": {
         "hotkey_type": "shortcut",
         "do": "win"
}
```

Thanks a lot.

braoult, 2019-04-27:
Hi,

I can confirm it works. Including japanese mappings.

JSubelj, 2019-05-04:
Hey,

I'm very glad to see someone taking an interest in my implementation. The reason for not including those event codes is that they are not region specific. On Slovenian keyboards KEY_Z event in reality corresponds to character "y" and vice versa. This would lead to a lot of confusion and significantly reduce the user experience. 

I will think about adding a `shortcut` like `hotkey_type` where you can input direct Linux input event codes, but sadly in not so near future because I have a lot of stuff going on in my life right now and sadly don't have the time to expand the code. 

If you did already add the Japanese keyboard mappings to char_uinput_mapper.py feel free to open a Pull request and I will gladly merge your code, so the community can benefit from it :)

Again thank you very much for taking an interest and I hope you have a good experience using my driver.

suabo, 2022-11-09:
I like the idea of mapping the gkeys directly to uinput key codes. I would call the new hotkey_type simply `mapping`.

suabo, 2022-11-14:
I just added the new hotkey_type **uinput** into a [feature branch](https://github.com/JSubelj/g910-gkey-macro-support/tree/hotkey_type-uinput). I did the smallest amount of documentation on this. So there is definitely room to improve. Maybe you can test it and tell me if it fits your needs.

suabo, 2023-09-10:
The requested feature got implemented with version 0.3.0.

closed 2023-09-10 by suabo