JSubelj/g910-gkey-macro-support #14: [Feature] 3 config files for the 3 M keys

Sebiann, 2019-05-04:
so i was wondering if it is Possible to add a config for each M key, cause i would like to be able to switch profile. so like m1-config.json and m2 and m3.

JSubelj, 2019-05-04:
Hey,

first I would like to say thank you for using my driver. 

This could definitely be possible and I will try to put it in my next release (but sadly probably the next release won't be any time soon because I have a lot of stuff going on in my life and currently don't have the time to expand the code). I thought about this kind of functionality from the very beginning but because I don't need it I didn't implemented it. 

If anyone would like to implement this feature please contact me and I will explain how to do it (its quite simple if you know a bit about programming :)).

In the mean time you can "hack" around this thing by creating a script that changes `/etc/g910-gkeys/config.json` with a diffrent config.json. The driver will pick up on the changes and you will have different functionalities on the GKeys.

If you have any questions please ask and I will gladly try to help you.

Sebiann, 2019-05-05:
Depending on what u mean with a bit of programming, i might be able to look in too it, ive been studying js and python lately. I will once read through the code and see if i come up with an idea.

JSubelj, 2019-05-05:
I'm very happy to hear that. My idea is to have a directory with 3 different config bindings and swapping the original config.json with the binding of the Mkey pressed. This also has to be backwards compatible with the old config but I think that shouldn't be a problem. 

We can then build on this to additionally back lit the MKeys with commands to [g810-led](https://github.com/MatMoul/g810-led) program.

The change can also be indicated to the user by using `notify-send` command. 

I'm very glad to see someone taking an interest in my project. If you have any additional questions you can ask them here or send me an email on: jan(dot)subelj010(at)gmail(dot)com  (With a properly named subject so it dosen't go to my spam folder :P)

braoult, 2019-05-06:
Hi,

I would not push on that system. I would prefer M-keys to be able to handle "normal keyboard event" (if possible), and their behavior not hard-coded as for example "switch to g910-gkey configuration [1-3]".

What I mean, the ideal (for me) would be to be able, with a command-line,  to switch such configurations. This would be phase 1.
And later, phase 2, if possible, to be able to map the M-keys as the G-keys are. So we could perform more sophisticated actions than just switching the G-keys mapping.

In two words, for me, being able to switch G-keys configuration and mapping the M-keys are two different subjects.

Correct me if what I say is nonsense :-)

Sebiann, 2019-05-06:
What u say makes sense, I think for the start your idea would be the easiest, but im not sure if its the best idea, im gonna test some stuff out this evening.

JSubelj, 2019-05-06:
@braoult I do agree with you. Probably the best implementation would be to have one separate config for the MKeys so that a more advanced user like yourself could change the behavior of MKeys the same way as with GKeys. But it should default to switching profiles as that is the intended behavior.

@Sebiann If you have a better idea please share so we can comment and agree on it.

braoult, 2019-05-06:
Yes, having a default config to change profile looks perfect (if changing
profile is scriptable, it is just the same as "run" function, therefore no
difference with a more general configuration).

MR-R080T, 2019-12-26:
So I have got mapping of M keys up and running like regular keys and I am going to work on the next few days when i get a chance to get profile switching working, I am loving the idea above of having them available to map to other mappings, i was thinking of adding a new hotkey type and that being switch profile, so in theory it could even be called by any G key. A crazy idea i had if i had time to implement would be changing profiles with dbus, programs could change their profile automagically?

MR-R080T, 2019-12-26:
This is now done, ive made a pull request

suabo, 2022-10-26:
I also liked the idea of having multiple profiles for the gkeys so i took my own shot on this topic and added profile switching without the need of swapping config files and having multiple config files to edit. I also figured a way out to send desktop notifications.
Thx @MR-R080T  for the great starting point. I used his swap-config branch to get going.

closed 2022-11-09 by suabo