JSubelj/g910-gkey-macro-support #17: [Question] Possibility of using this  with G613

aryasenna, 2019-06-01:
So Logitech G613 also happen to have G1-G6 keys and I was wonderingf if we can adapt this code.

I tried to run it as is, but got the following error:

```
019-06-01 20:22:10,934 [DEBUG] {lib.keyboard_initialization.usb_and_keyboard_device_init} - usb dev: not found. retrying with other product id
```
Probably because this program is hardcoded to G910 product ID, but I noticed actually it is bruteforcing stuff until it got some correct device ID:

https://github.com/JSubelj/g910-gkey-macro-support/blob/5caa954019a835906f80950d1f21877e9b1b7405/lib/keyboard_initialization/usb_and_keyboard_device_init.py#L26

Can @JSubelj  perhaps comment if this is even possible?

JSubelj, 2019-06-02:
Hey,
thank you for taking an interest into my driver. 

It's not really bruteforcing it's just switching between USB_PRODUCT ids. One is for Orion Spark and one is for Orion Spectrum. 

About the G613. I'm not really sure it will work because of the hexvalues that the keyboard sends are probably very different then on G910 (could be wrong). First thing you should do is try to find usb product and usb vendor ids (most likely usb vendor id is the same). Wireshark with USB sniffer will probably help you with that. Then just modify the code so that USB_PRODUCT[0] points to your usb product id. If it connects try it out.

If it works then great. If not, look at the log files if you have any warnings in style of: `[hexval] no match`. That means that G613 is sending different hex codes when GKey is pressed. If so, collect the hex values and replace them here: [command_bytearray.py](https://github.com/JSubelj/g910-gkey-macro-support/blob/5caa954019a835906f80950d1f21877e9b1b7405/lib/data_mappers/command_bytearray.py). 

There is also a possibility that your GKeys are bind to FKeys. You will have to disable this binding first (my driver does disable them but only for G910 idk. if it also works on G613). You can do this with [g810-led](https://github.com/MatMoul/g810-led) utility.

If you have any questions please ask right away. If you have any success then fork the code and submit a pull request so others can use the driver on their G613 too :).

aryasenna, 2019-06-02:
Hey @JSubelj,
Thanks for your quick and detailed reply, appreciate that.

So first, I'm not sure if USB sniffer+Wireshark is needed to get the USB Product ID. Is the good old `lsusb` not enough? (forgive my ignorance if you are talking about something different).

`$ lsusb`
`Bus 003 Device 007: ID 046d:c53d Logitech, Inc. USB Receiver`

I then change the PID on https://github.com/JSubelj/g910-gkey-macro-support/blob/5caa954019a835906f80950d1f21877e9b1b7405/lib/keyboard_initialization/usb_and_keyboard_device_init.py#L19

to 
` USB_PRODUCT = (0x0c32b, 0xc53d) `

then start `# g910-gkeys` again:

```

2019-06-02 23:59:26,400 [INFO] {launcher} - ------------------------------------------------------------------------------------
     4 2019-06-02 23:59:26,400 [INFO] {launcher} - ----------------------STARTED g910-keys-pid:18343-----------------------------------
     5 2019-06-02 23:59:26,400 [INFO] {launcher} - ------------------------------------------------------------------------------------
     6 2019-06-02 23:59:26,400 [DEBUG] {lib.data_mappers.config_reader} - Reading config from /etc/g910-gkeys/config.json
     7 2019-06-02 23:59:26,400 [DEBUG] {lib.data_mappers.config_reader} - using config: {'keyboard_mapping': 'si', 'g1': {'hotkey_type': 'typeout', 'do': 'Its WORKING!!!%%'}, 'g2': {'hotkey_type': 'nothing', 'do': ''}, 'g3': {'hotkey_type       ': 'nothing', 'do': ''}, 'g4': {'hotkey_type': 'nothing', 'do': ''}, 'g5': {'hotkey_type': 'nothing', 'do': ''}, 'g6': {'hotkey_type': 'nothing', 'do': ''}, 'g7': {'hotkey_type': 'nothing', 'do': ''}, 'g8': {'hotkey_type': 'nothing       ', 'do': ''}, 'g9': {'hotkey_type': 'nothing', 'do': ''}}
     8 2019-06-02 23:59:26,400 [DEBUG] {lib.keyboard_initialization.usb_and_keyboard_device_init} - gathering usb device
     9 2019-06-02 23:59:26,412 [DEBUG] {lib.keyboard_initialization.usb_and_keyboard_device_init} - usb dev:     INTERFACE 1: Human Interface Device ====================
    10      bLength            :    0x9 (9 bytes)
    11      bDescriptorType    :    0x4 Interface
    12      bInterfaceNumber   :    0x1
    13      bAlternateSetting  :    0x0
    14      bNumEndpoints      :    0x1
    15      bInterfaceClass    :    0x3 Human Interface Device
    16      bInterfaceSubClass :    0x1
    17      bInterfaceProtocol :    0x2
    18      iInterface         :    0x0 
    19       ENDPOINT 0x82: Interrupt IN ==========================
    20        bLength          :    0x7 (7 bytes)
    21        bDescriptorType  :    0x5 Endpoint
    22        bEndpointAddress :   0x82 IN
    23        bmAttributes     :    0x3 Interrupt
    24        wMaxPacketSize   :   0x20 (32 bytes)
    25        bInterval        :    0x1
    26 2019-06-02 23:59:26,413 [INFO] {lib.keyboard_initialization.usb_and_keyboard_device_init} - Trying to disable mapping
    27 2019-06-02 23:59:26,413 [DEBUG] {lib.keyboard_initialization.usb_and_keyboard_device_init} - Sending disable
    28 2019-06-02 23:59:26,413 [WARNING] {lib.keyboard_initialization.usb_and_keyboard_device_init} - Disable mapping error: [Errno 32] Pipe error

```

Okay, It went past the `usb dev: not found` error, but then it got stuck with error in the GtoFkeys-binding disable function. The returned error, as you see, is  some low level error from PyUSB/libusb :/

I have also tried g810-led utilities as you suggested, naturally G613 is not among the supported keyboards, so I went ahead and use the `-tuk` (test untested keyboard) flags:

 `$ g513-led -dv 046d -dp c53d -tuk 3 -gkm 1 `
`Matching or compatible device not found !`

I tried with all possible protocols (1-3) and also with other keyboard-specific binary (`g610-led`, `g513-led`, `g810-led`, `g910-led`), but it's all no go :(

Few things I noticed
1. G613 uses this wireless USB dongle (the Lightspeed one). Maybe there is an extra abstraction layer/different protocol between the dongle and the keyboard (thus the error from PyUSB/libusb).
2. On Windows Logitech G-Hub program, I don't see a different UI/tool is used to disable G to F keys binding for G613 as compared when using other keyboard. Of course they can do it differently under the hood and I guess it's possible to know what's happening using sniffer as you suggested.

Gee, I hate this when Linux is always a second class citizen when it comes to consumer hardware support!

Any suggestions are welcome, though.

JSubelj, 2019-06-02:
Yeah sry `lsusb` is definitely right for the job.

Because you're using a wireless keyboard then there probably is some sort of abstraction on top of Logitech's hardware protocol.

There is a possibility that you don't really need GKey to FKey binding disabled. Are your GKeys bind to FKeys even? If not then maybe you can get around the FKey disabling in my software by commenting the code out.

And yeah I feel you. I'm almost certain Logitech won't support Linux until something really big happens gaming wise. Let's hope Google Stadia will draw their attention. I've seen a post on Logitech forum from 2013 that almost begs for Linux support and they said they will implement it but yeah...

aryasenna, 2019-06-03:
Yeah I have tried commenting out the GtoFkeys-binding disable function, but no go.
Unfortunately the G keys are definitely bound to the F keys.

I checked the key hex value using `xev`; G(1-5) and F(1-5) are identical.

JSubelj, 2019-06-03:
I see. Then sadly I can't really help you. My GtoFkeys disable function is ported from g810-led. I think you will need to reverse engineer the protocol (by looking at packets sent by windows utility to the keyboard when windows utility is turned on). You could also ask for help on (g810-led)[https://github.com/MatMoul/g810-led]. Maybe they will find a way to direct you in the right direction. 

If you figure out how to disable GtoFKeys binding then reopen this issue and I'll try to help you further. Good luck :)

closed 2019-06-03 by JSubelj