JSubelj/g910-gkey-macro-support #18: on printout @ is replaced with "

c0ff3e, 2019-06-14:
So i added my email to G2 key and when using it @ is replaced with "

JSubelj, 2019-06-14:
What kind of keyboard layout are you using and what kind of keyboard layout is set in config.json?

c0ff3e, 2019-06-14:
swedish 
and have now tested both en and si .

JSubelj, 2019-06-14:
Do swedes have @ on the same place in the layout as si or en? If not you have to either implement the swedish keyboard layout or change english or slovenian keyboard layout to your desire here [char_uinput_mapper.py](https://github.com/JSubelj/g910-gkey-macro-support/blob/master/lib/data_mappers/char_uinput_mapper.py).

c0ff3e, 2019-06-14:
altgr+2 
i have tested chaning that file  adding both se as language and modifying the en.

BTW the documentation dont say anything about the lang mapping in config file.. could be a litle more clear .

c0ff3e, 2019-06-14:
update: i did edit my local checkout file not system file.
i will send you a full swedish layout so you can add it .

JSubelj, 2019-06-14:
So if I understand you correctly it works now for you? If it does please open a pull request and I will merge it.

I will probably extend the wiki to include additional information how to add keyboard layouts.

Thank you for your contribution.

c0ff3e, 2019-06-14:
I will make a pull next week.,. thanxs for the help ( yes its working now )

closed 2019-06-14 by c0ff3e