JSubelj/g910-gkey-macro-support #22: [Feature] Add support for additional hotkey types

loredous, 2020-02-01:
Having the ability to do simple macros and run functions is nice, but sometimes it just doesn't cut it when you really want to make powerful hotkeys.

Initial support should be added with the ability to provide a python one-liner, or possibly even load full python functions, and have the hotkey perform that function, and write any string output that is returned.

loredous, 2020-02-01:
I will be working on this and should have an initial pull request by mid Feb.

braoult, 2020-02-01:
I do not understand... If the keys are mapped to F-keys, this can be done on system side, no need for driver... I really think that minimal is best: remove "replace fkey", useless, just make them normal keys.

I disagree to send a string on a G-key, useless. Send a F- signal, which can be handled everywhere. This includes sending string.

loredous, 2020-02-01:
> I do not understand... If the keys are mapped to F-keys, this can be done on system/application side, no need for driver... I really think that minimal is best: remove "replace fkey", useless, just make them normal keys.
> 
> I am puzzled to have G-keys able to send a string, it looks useless (working on one application only).
> Sending a F- signal (F13, F14, etc...), which can be managed everywhere, can be easily handled.
> 
> What is the point to have "G2" to write "ls -lart" instead of it sending "F14", then managing "F14" on application side ?
> 
> F14 could be mapped to something in emacs, something else in LibreOffice, etc...
> Sending a string just works for one application only (shell OR emacs OR LibreOffice etc...).

I'm not quite sure I understand the objection here. This seems like a logical extension of the existing functionality of this tool. This allows for a G-key to send a dynamic string, even when using underlying applications that do not support hotkey functionality. For example, I would like to have a hotkey that can insert a date/time stamp, regardless of what application I am in. With this functionality, I can have a hotkey that is able to get the value for `datetime.now()` and output it in my desired format. To do this currently would require that I map the G-key to a function key, and then run a separate application to handle the pressing of that function key.

It does not take away the ability to have G-keys mapped to F-keys, only adds functionality that requires external applications to achieve currently.

braoult, 2020-02-02:
hmmm... OK. But this makes the code more complex, for no real added value, as it would be a system-wide setting, not a user setting.

Usually, every user defines the keyboard behavior, there are settings for that.

Hardcoding code at system level looks wrong (compared to simple f-key mapping), as it is impossible to change anything. Code mode complex, no added value, restrictions on user settings, this is what I see...

JSubelj, 2020-02-05:
Hey, thanks for your feature. On one side I agree with @braoult, that this can also be made on system level and implementing this feature is kinda against [Unix philosophy](https://en.wikipedia.org/wiki/Unix_philosophy). Also all commands are executed with superuser permission which could raise all hell. 

On the other side I see this as a good and easy way to use typeout system of the driver and especially using it in CLI only interface.

Because the code addition is rather small and it does not complicate the driver by that much or slow it down (also this is still a user side driver written in python of all languages) I will probably allow the pull request when I test it out when I'll find the time.

closed 2020-02-05 by JSubelj

braoult, 2020-02-05:
Still system-wide output string, right ? I will stay with F-XX mapping, every user choosing what they are for, instead of having a date inserted, which will only interest one and make the G-keys useless to someone who prefers "hyper" key or launching a music player :)

JSubelj, 2020-02-06:
Hey @braoult, this issue/pull request is about adding the ability of running python code directly when pressing GKeys. You can also see the edited code here (https://github.com/JSubelj/g910-gkey-macro-support/pull/23). This issue/pull request does not change the behavior of the driver. This just expends it. So everything you are doing now with the driver will also work in the future.

The issue with global/local config you described is already present  in the current release. It will probably be fixed with conf file in $HOME.

braoult, 2020-02-09:
I think it does: You can run any code/command with F-keys... Python is not for everybody, only python programmers can do... This driver is also for children and grandma, using same keyboard, one for gaming, one to launch "skype". I mean end-users. If we keep oriented "python-oriented", it will fail for both targets, and keep useful only from python developers for python developers...

I really believe that matching G-keys to F-keys by default could help "normal" user to associate that F-key to anything. Including python code. or perl. or bash. or firefox. or chat with grand-childs.

JSubelj, 2020-02-10:
You are absolutely right when it comes to usability. It is meant for everyone. But this addition to the driver does not take any usability from anyone, only extends it to someone else. Everything will stay the same as it was before. You will not see any difference in the driver functionality or performance. It's like having an air conditioning in the car. If you don't want to use it, don't use it. It does not change anything about the car it is only an addition to it.

I don't really understand what you mean with matching GKeys to FKeys by default. This is already done by default and you don't need my driver for that. My driver disables this automatic mapping because G1 key pressed w/o the driver can not be differentiated from F1 key. Any user can enable G to F key mapping in config.json or bind anything if they so please.

This driver is written in python so it is python oriented. It is also user-spaced so it does not have anything to do with the kernel. This will not change until someone will port the whole thing to C and into a kernel module. Implementing python code execution does not have any additional performance cost because the whole thing is written in python. 

But yeah I think I do understand you. You wan't to say that the core of the driver is mapping signals sent from the keyboard to signals that OS understands. I agree with you. But this does also mean that I should throw out typeout and run config options as well. But I won't because I use them on daily basis.

In the end is like this. If the user requests a feature, adds it, pushes it, the feature does not break any backwards compatibility or does not introduce any significant changes in performance it will most likely be added into the driver. In this way the driver is easier to use by everybody because it covers a bigger specter of users, it becomes more popular, gets more dev working on it etc. etc.

PS: About children or grandmas, a lot of home appliances have additional features that grannys or children do not use. This does not mean that they should be thrown out, because grannys and children are not the only users of these home appliances.

braoult, 2020-03-23:
_In the end is like this. If the user requests a feature, adds it, pushes it, the feature does not break any backwards compatibility or does not introduce any significant changes in performance it will most likely be added into the driver. In this way the driver is easier to use by everybody because it covers a bigger specter of users, it becomes more popular, gets more dev working on it etc. etc.

PS: About children or grandmas, a lot of home appliances have additional features that grannys or children do not use. This does not mean that they should be thrown out, because grannys and children are not the only users of these home appliances._

Grandma will never push any change in python code, nor write a "date" function.

But she could map F20 to "launch radio", while I would prefer "re-indent my Emacs buffer", and "launch radio" while not in Emacs. This is the point of F-keys: application side, not system-side.

AFAIU, in your case, you may use the F-keys to emit what you want (a date or whatever), even if a terminal and a LibreOffice or Emacs or vi could want different mappings (emacs understands Hyper key, vi or LibreOffice do not).
I just do not understand why Gkey->Fkey is not the default, allowing a full user configuration, on application side.

Python is not for everybody. An F-key is.