JSubelj/g910-gkey-macro-support #26: [Issue] F keys appear to be case sensitive

ajpirciovs, 2020-04-13:
**Describe the bug**
When setting up shortcuts with F1-12 keys, you must specify "F#" and not "f#"

**To Reproduce**
Steps to reproduce the behavior:
edit: /etc/g910-gkeys/config.json, use "f#" key with type shortcut

**Expected behavior**
should be case insensitive, or example docs updated (example shows "alt+f4", adding "alt+F4". Also consider adding a comment here to make sure noobs like me are paying attention)

ihkN, 2020-05-08:
I'd suggest to make lower or upper case all chars in the string before using it.

suabo, 2022-11-09:
Thanks for reporting the Problem.  
I will see if I can allow for case insensitive behavior at the mapping. Other ways I'll add a comment to the documentation.