JSubelj/g910-gkey-macro-support #27: [Issue] Service does not install on linux-mint 19.3

ajpirciovs, 2020-04-13:
**Describe the bug**
After running installer.sh a service file exists in the installation directory but is NOT copied over to /etc/systemd/system. The documentation refers to /usr/lib/systemd/system/g910-gkeys.service which is not the correct folder in this distro 

**To Reproduce**
Steps to reproduce the behavior:
1. Go thru install steps on linix-mint 19.3 (or probably ubuntu as well)
2. See that service file is not present where indicated

**Expected behavior**
Should detect system service file location path and copy appropriately

**Desktop (please complete the following information):**
 - OS: Linux Mint 19.3 Tricia
 - Kernel version: Linux Mint 19.3 Tricia
 - python version: 3.7.4

ajpirciovs, 2020-04-13:
I would like to add that this is a great utility and I thank you much!

braoult, 2020-06-17:
This one should be fixed after #36 . I think this case can be closed.

closed 2020-06-18 by JSubelj