JSubelj/g910-gkey-macro-support #37: [Issue] Uninstall instructions are wrong (incomplete).

braoult, 2020-06-18:
Uninstall instructions rely on "files.txt", which contains files, not directories.

It means (in case of Debian), the installed files in /usr/local/lib/python3.8/dist-packages/g910_gkeys-0.2.4-py3.8.egg are removed, but not the directory itself.

I would suggest to replace files.txt with a list of files *OR* directories when applicable (for instance file for systemd service file, but directory for /usr/local/lib/python3.8/dist-packages/g910_gkeys-0.2.4-py3.8.egg.

**To Reproduce**
follow instructions to uninstall.
/usr/local/lib/python3.8/dist-packages/g910_gkeys-0.2.4-py3.8.egg is still there with subdirs (and some __pycache__ dirs added).

**Expected behavior**
/usr/local/lib/python3.8/dist-packages/g910_gkeys-0.2.4-py3.8.egg  should have disappeared.

**Desktop (please complete the following information):**
 - OS: XUbuntu 20.04.
 - Kernel version: 5.4.0-37-generic
 - desktop enviroment: XFCE
 - g910-gkeys version: 0.2.4
 - python version: 3.8.2

suabo, 2022-11-09:
On my last builds i got only 3 files in files.txt:
```
/usr/local/bin/g910-gkeys
/usr/local/lib/python3.8/dist-packages/g910_gkeys-0.2.5-py3.8.egg
/lib/systemd/system/g910-gkeys.service
```
Maybe you can install the latest version and double check it?
I also tried to uninstall with `pip uninstall g910-gkeys` which also removes `/usr/local/lib/python3.8/dist-packages/g910_gkeys-0.2.5-py3.8.egg` so in uninstall.sh it is redundant.

suabo, 2024-05-27:
Since install and uninstall is handled by pip now (no more files.txt), there is no more problem with left files on filesystem.

closed 2024-05-27 by suabo