JSubelj/g910-gkey-macro-support #46: Add Python3 Distutils Enhancements

Axellarator, 2021-01-10:
Add python3-setuptools if missing **setup** error during installaton pops up.
sudo apt install python3-setuptools

suabo, 2022-11-09:
Thank you, I will add this somewhere.

Torpedro1978, 2023-05-22:
I get this error after 

sudo apt install python3-setuptools

`Running pyusb-1.0.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-7rufsmy6/pyusb-1.0.2/egg-dist-tmp-pw6uo9mp
/usr/lib/python3/dist-packages/setuptools/dist.py:775: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
  warnings.warn(
warning: no files found matching 'ReleaseNotes.rst'
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
`

suabo, 2023-09-10:
@Torpedro1978 Thanks for your feedback!
> I get this error after
>
> sudo apt install python3-setuptools
>
> `Running pyusb-1.0.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-7rufsmy6/pyusb-1.0.2/egg-dist-tmp-pw6uo9mp [...]

**First of all this is not an error, it's just a warning.** 
The warning seams to be related to pyusb. We never updated the required version. It's likely that the warning goes away with in a newer version of pyusb. I will look into that and maybe set a higher version for pyusb in the dependencies of the driver.

closed 2024-05-27 by suabo