JSubelj/g910-gkey-macro-support #49: [Issue] Running pacmd commands (switching sound device by command) will not succeed

Simon1511, 2021-06-07:
**Describe the bug**
If you set the hotkey_type of any of the G-Keys to "run" to make it run a command, the command "pacmd set-card-profile profileIndex" (where profileIndex is the index of the sound card you want to switch to) will not succeed (i.e. it will do nothing). The command is supposed to switch sound devices, e.g. from speaker to headphones.

**To Reproduce**
Steps to reproduce the behavior:
1. Open /etc/g910-keys/config.json in any text editor
2. Set hotkey_type to "run" for your preferred G-Key
3. Add "pacmd set-card-profile profileIndex" to the "do" part of your G-Key (Replace "profileIndex" with the index of the sound card you want to switch to).
4. See that nothing will happen

**Expected behavior**
The command is supposed to switch sound output and input devices. In my case, it should switch from speakers to my headset. Running the command directly in a terminal window works fine.

**Screenshots**
None

**Additional info**
I have also tried putting the pacmd command into a shell script in /usr/local/bin and then add the shell script to the "do" section. Didnt work either.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 (same thing happened earlier on 18.04)
 - Kernel version: 5.8.0-55-generic
 - desktop enviroment: Gnome 3.36.8
 - g910-gkeys version: latest
 - python version: 3.8.5

Simon1511, 2021-06-07:
Thanks for the answer, but from what I read in the Wiki its only not possible to run _Graphical_ programs, that means with a GUI, right?

Here, pacmd is a commandline only tool, so there shouldn't be any issues with it? Or am I wrong?

JSubelj, 2021-06-08:
When I was writing the wiki, I centered around GUI programs, but the issues remain. The problem is that the driver is running as a superuser. Superuser does not share env vars, programs and other stuff with normal users.  That's why your program probably doesn't work. 

Run command is dangerous because it gives sudo privileges to every program it runs and is a security concern. It is also finicky and does not properly report if anything goes wrong. Maybe in the future I'm gonna axe it all together.

Simon1511, 2021-06-08:
Ah ok, thanks for the detailed explanation.
Will close the issue then and find another solution.

closed 2021-06-08 by Simon1511

braoult, 2021-06-08:
@Simon1511, my preferred setup is to configure my G-keys as F-keys only :
```
{
    "__comment": "I was unable to use F20, even xev does not show the keycode. It just mutes/unmutes sound, same as the G910's <mute> button",
    "keyboard_mapping": "en",
    "g1": {
        "hotkey_type": "shortcut",
        "do": "F13"
    },
    "g2": {
        "hotkey_type": "shortcut",
        "do": "F14"
    },
    "g3": {
        "hotkey_type": "shortcut",
        "do": "F15"
    },
    "g4": {
        "hotkey_type": "shortcut",
        "do": "F16"
    },
    "g5": {
        "hotkey_type": "shortcut",
        "do": "F17"
    },
    "g6": {
        "hotkey_type": "shortcut",
        "do": "F18"
    },
    "g7": {
        "hotkey_type": "shortcut",
        "do": "F19"
    },
    "g8": {
        "hotkey_type": "shortcut",
        "do": "F21"
    },
    "g9": {
        "hotkey_type": "shortcut",
        "do": "F22"
    }
}
```
Then, I define shortcuts for those F-keys as I could do with the usual ones:
Either at WM level (for instance to raise Emacs), **this is what you want,** or at application level.

**Warning**: F13 and next ones have predefined settings, this is why the configuration below is mandatory (if you want to use "Shift F13" for example).

**DISCLAIMER**: below is only for `Xorg`. Many distributions are moving to `Wayland`, I did not (yet !) study if there is an easy way to convert Xmodmap to an equivalent in Wayland, the few I found looks like it is a nightmare, so I always prefer to work on it "tomorrow" :-)

Note: In the setting below, `G5` becomes `Hyper` key - another modifier such as `Shift`, `Control`, `Meta` (alt), and `Super` (windows) - in my .Xmodmap, and `Caps Lock` becomes a new `Control` key, this is maybe not what you want.

My full .Xmodmap is :
```
! .Xmodmap - keyboard mapping
!
! br, 2015-2020
!

!!!!!!!!!!!!!! Original modifier keys
! $ xmodmap -pke | grep Control gives:
! keycode  37 = Control_L ISO_Prev_Group Control_L
! keycode 105 = Control_R ISO_Next_Group Control_R ISO_Next_Group
! keycode 195 = Control_L Control_L Control_L Control_L

!!!!!!!!!!!!!! Super keys
! Map "Windows right" to Meta_R and "Menu right" to Super_R

keycode 134 = Meta_R NoSymbol Meta_R
keycode 135 = Super_R NoSymbol Super_R

! necessary on my system, for some reason original modifier for mod4 is:
! mod4        Super_L (0x85),  Meta_R (0x86),  Super_L (0xce),  Hyper_L (0xcf)
clear mod4
add mod4 = Super_L Super_R

!!!!!!!!!!!!!! Caps Lock
! Caps Lock becomes a new Control_L key
keycode 66 = Control_L
clear Lock
add Control = Control_L

!!!!!!!!!!!!!! Japanese standard keyboard
! Muhenkan mapped to Hyper_L
! menu key mapped to Hyper_R

! keysym Muhenkan = Hyper_L Muhenkan
! keysym Menu = Hyper_R Menu
! clear mod4
! add mod4 = Super_L Super_R
! clear mod3
! add mod3 = Hyper_L Hyper_R

!!!!!!!!!!!!!! Logitech G910 keyboard
! Note:
! F11-F20 have synonyms L1-L10
! F21-F35 have synonyms R1-R15
! See: /usr/include/X11/keysymdef.h
!
! original F13, F14, etc... mapping
! F13 = keycode 191 = XF86Tools NoSymbol XF86Tools
! F14 = keycode 192 = XF86Launch5 NoSymbol XF86Launch5
! F15 = keycode 193 = XF86Launch6 NoSymbol XF86Launch6
! F16 = keycode 194 = XF86Launch7 NoSymbol XF86kLaunch7
! F17 = keycode 195 = XF86Launch8 NoSymbol XF86Launch8
! F18 = keycode 196 = XF86Launch9 NoSymbol XF86Launch9
! F19 = keycode 197 = (nothing)
! F20 = keycode 198 = XF86AudioMicMute NoSymbol XF86AudioMicMute
! F21 = keycode 199 = XF86TouchpadToggle NoSymbol XF86TouchpadToggle
! F22 = keycode 200 = XF86TouchpadOn NoSymbol XF86TouchpadOn

! we map the Gkeys (G1-G9), to F13-F22,
! Except G5, that becomes Hyper_L

! G1
keycode 191 = F13 F13 F13

! G2
keycode 192 = F14 F14 F14

! G3
keycode 193 = F15 F15 F15

! G4
keycode 194 = F16 F16 F16

! G5
! it becomes Hyper_L
keycode 195 = Hyper_L NoSymbol Hyper_L
clear mod3
add mod3 = Hyper_L

! G6
keycode 196 = F18 F18 F18

! G7
keycode 197 = F19 F19 F19

! G8
keycode 199 = F21 F21 F21

! G9
keycode 200 = F22 F22 F22
```