JSubelj/g910-gkey-macro-support #51: [Feature] Support for holding down keys in shortcuts

EisbarGFX, 2021-08-11:
One thing I wish would be possible, which I don't know if it is *possible* with this software, is the option to hold down a shortcut key. Specifically, this would be useful for alt-tabbing. Often, I use alt+tab and hold down alt, which lets me use the arrow keys or mouse to select apps to switch to other than the last app.

Don't even know if this is possible to implement, so I don't really have any ideas for alternatives... I do realize that the arrow keys aren't supported as shortcuts, so that would be something that further blocks this feature.

braoult, 2021-08-11:
I am not sure to understand what you mean, but I use G5 as an "`Hyper`" key, which is another modifier, such as `Shift`, `Control`, `Meta` (alt), and `Super` (windows). To do so, I map G5 to a F-key in `config.json`, and create the `Hyper` modifier in `~/.Xmodmap` as following:

**/etc/g910-gkeys/config.json:**
```
{
    [...]
    "g5": {
        "hotkey_type": "shortcut",
        "do": "F17"
    },
    [...]
}
```
**~/.Xmodmap:**
```
[...]
! G5 becomes Hyper_L
keycode 195 = Hyper_L NoSymbol Hyper_L
clear mod3
add mod3 = Hyper_L
[...]
```

Is it what you want ?

suabo, 2022-11-09:
If you bind alt+tab to a shortcut and keep it pressed the gkey act like you press alt+tab individually. On my ubuntu it switches the tabs very fast (not useful) if you keep both keys pressed. You want the shortcut to hit alt+tab and release tab whiles you still pressing the gkey, right? That way you could navigate through the tabs like you described.

To implement this you would need to define a special rule for this shortcut. Then you could release the tab key right after the gkey was pressed. I don't think this is a feature which is useful for everybody so I will close the issue.

closed 2022-11-09 by suabo