JSubelj/g910-gkey-macro-support #52: [Issue] "ModuleNotFoundError: No module named 'usb'"

FabiSahne, 2021-08-13:
**Describe the bug**
After following the detailed instructions the command 'systemctl status g910-gkeys' gives following output:
https://pastebin.com/9iF7zm0h

**To Reproduce**
Steps to reproduce the behavior:
1. Install
2. systemctl status g910-gkeys
3. See error

**Additional info**
 - pip, pyusb 1.0.2, python-uinput 0.11.2 are installed
 - logs from /var/log/g910-gkeys.log are empty

**Desktop:**
 - OS: Pop!_OS 21.04
 - Kernel version 5.11.0-7620-generic
 - python version 3.9.6

suabo, 2022-10-31:
It looks like the driver can't load the pyusb package. First thing i would try is to use python3.8.
I'm using python 3.8.10 and never tested it on 3.9.

If this don't work try to uninstall pyusb and python-uinput via pip and install it via distribution package:
`apt install python-usb python-uinput` or `apt install python3-usb python3-uinput`

suabo, 2023-09-06:
I now tested with python version 3.9.18 and everything works fine.
Can you download the newest version (0.3.0) and try again?

closed 2024-05-27 by suabo