JSubelj/g910-gkey-macro-support #54: [Question]

akovia, 2021-09-16:
I can't find any information on g815 support, and since I can't seem to get it to work I guess it's not supported.

Are there any plans to support the G815 in the future? I am desperate to use my 5 G-keys as shortcuts, but can't find support anywhere.

Thanks for any information at all!

JSubelj, 2021-09-16:
Hey, thanks for the question. G815 keyboard is not supported because I don't have one and its hard to develop support without the hardware. I guess you can try your luck and develop your own based on my code, because I doubt there is a lot of difference. Good luck :)

closed 2021-09-16 by JSubelj

akovia, 2021-09-16:
Appreciate the fast reply!

I believe developing this would be beyond my current abilities, but it's good to know I wasn't overlooking something so I can go ahead and lay it to rest.

Cheers!

EinWildesPanda, 2021-09-25:
I think it's reasonable to assume that it should work by updating the product ids the driver searches for (unless I misunderstood sth). Think I have a spare G815 laying around somewhere, so I'll test it sometime in the future.

TobiPeterG, 2022-04-20:
> I think it's reasonable to assume that it should work by updating the product ids the driver searches for (unless I misunderstood sth). Think I have a spare G815 laying around somewhere, so I'll test it sometime in the future.

Hey there,
have you made some progress on this?

EinWildesPanda, 2022-04-21:
> > I think it's reasonable to assume that it should work by updating the product ids the driver searches for (unless I misunderstood sth). Think I have a spare G815 laying around somewhere, so I'll test it sometime in the future.
> 
> Hey there, have you made some progress on this?

Yes, but not quite the way I planned. I wrote a less complicated macro manager which requires you to remap your gkeys with libratbag. I'm going to clean it up a little and release it in the following days.

TobiPeterG, 2022-04-22:
> Yes, but not quite the way I planned. I wrote a less complicated macro manager which requires you to remap your gkeys with libratbag. I'm going to clean it up a little and release it in the [...]

Ok, thank you very much! :)

EinWildesPanda, 2022-04-23:
> Ok, thank you very much! :)

I just published [the repository](https://github.com/EinWildesPanda/Logitech-Linux-macros). I believe I didn't forget anything, though I'm not 100% sure.