JSubelj/g910-gkey-macro-support #55: [Feature] Logitech G915 support?

hogar1977, 2021-11-23:
Hi there,
I use Arch Linux with Logitech G915 keyboard (wireless/lightspeed connection) and I installed your package from AUR, I enabled the service and tried to start it but it fails with the following report:

```
× g910-gkeys.service - Support for Logitech g910 keyboard gkeys
     Loaded: loaded (/usr/lib/systemd/system/g910-gkeys.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Tue 2021-11-23 17:51:52 CET; 2s ago
       Docs: https://github.com/JSubelj/g910-gkey-macro-support/wiki
    Process: 3426 ExecStart=/usr/bin/env g910-gkeys (code=exited, status=1/FAILURE)
   Main PID: 3426 (code=exited, status=1/FAILURE)
        CPU: 62ms

stu 23 17:51:52 dalibor-rogstrix env[3426]:     device = uinput.Device(uinput_all_keys.uinput_all_keys)
stu 23 17:51:52 dalibor-rogstrix env[3426]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 178, in __init__
stu 23 17:51:52 dalibor-rogstrix env[3426]:     self.__uinput_fd = fd or fdopen()
stu 23 17:51:52 dalibor-rogstrix env[3426]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 84, in fdopen
stu 23 17:51:52 dalibor-rogstrix env[3426]:     return _libsuinput.suinput_open()
stu 23 17:51:52 dalibor-rogstrix env[3426]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 70, in _open_error_handler
stu 23 17:51:52 dalibor-rogstrix env[3426]:     raise OSError(code, msg)
stu 23 17:51:52 dalibor-rogstrix env[3426]: OSError: [Errno 19] Failed to open the uinput device: No such device
stu 23 17:51:52 dalibor-rogstrix systemd[1]: g910-gkeys.service: Main process exited, code=exited, status=1/FAILURE
stu 23 17:51:52 dalibor-rogstrix systemd[1]: g910-gkeys.service: Failed with result 'exit-code'.
```

Is there anything I should do to make it work w this keyboard - don't know if the wireless part is whats tripping the whole thing. I would really like to use the G keys if I can. The keyboard button layout is essentially the same as on G910... Also does it support HR (hrvatski) layout :D?

jason-pollock, 2022-09-04:
@hogar1977 - have you made any inroads on this question? I'm also running a 915 and have the same questions. And the LED wave mode is driving me nuts.

suabo, 2022-11-09:
Look what i just stumbled over: [https://github.com/MatMoul/g810-led/blob/master/src/classes/Keyboard.cpp#L793](https://github.com/MatMoul/g810-led/blob/master/src/classes/Keyboard.cpp#L793)  
`data = { 0x11, 0xff, 0x0a, 0x2b, value };`
It looks like there is the byte code to activate gkeys on g915.
I can do the changes, but i need one of you to test it. Is one of you interested?

p5-f20w18k, 2023-01-12:
Did you update the code for this? I will test it

FnigePython, 2023-05-13:
Any updates on this? @suabo @JSubelj

suabo, 2023-09-06:
>= v0.3.0 will support the following keyboards:

- LogitechG915
- LogitechG910OrionSpectrum
- LogitechG910OrionSpark
- LogitechG815
- LogitechG710p
- LogitechG510

Can someone test it and tell me if it works?

jnk22, 2023-09-06:
Hi, I installed the current master branch and both [python-uinput](https://aur.archlinux.org/packages/python-uinput) and [python-pyusb](https://archlinux.org/packages/extra/any/python-pyusb).

Keyboard: **Logitech G915** (de) with USB cable plugged in
OS: Arch Linux

I have installed the current version using the `installer.sh` approach. As no default config has been created automatically, I have also created a **/etc/g910-gkeys/config.json** and replaced "en" with "de".

After that, I ran

```bash
sudo modprobe uinput
```

This error is what I get now in my logs:

```console
Started Support for Logitech g910 keyboard gkeys.
Traceback (most recent call last):
  File "/usr/bin/g910-gkeys", line 33, in <module>
    sys.exit(load_entry_point('g910-gkeys==0.3.0', 'console_scripts', 'g910-gkeys')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/g910_gkeys-0.3.0-py3.11.egg/cli_entry_point.py", line 28, in main
  File "/usr/lib/python3.11/site-packages/g910_gkeys-0.3.0-py3.11.egg/lib/g910_gkey_mapper.py", line 63, in main
  File "/usr/lib/python3.11/site-packages/g910_gkeys-0.3.0-py3.11.egg/lib/usb_device.py", line 52, in disable_fkey_to_gkey_binding
  File "/usr/lib/python3.11/site-packages/usb/core.py", line 1082, in ctrl_transfer
    ret = self._ctx.backend.ctrl_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 893, in ctrl_transfer
    ret = _check(self.lib.libusb_control_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
g910-gkeys.service: Main process exited, code=exited, status=1/FAILURE
g910-gkeys.service: Failed with result 'exit-code'.
```

I have not used this software before, so I am unsure if this is due to misconfiguration on my side or any other error.

suabo, 2023-09-17:
With v0.4.0 driver is using pyusb=1.2.1, since then the package is now compatible with python3.11.  
Can somebody else verify that it works with python3.11.
I successfully tested it with  3.8.10, 3.9.18 and 3.11.5.

jnk22, 2023-09-23:
Thanks for your update :)

I was able to install it via the AUR. It seems that the installation was succesful.

However, this is what I get when running `g910-gkeys` via terminal:

**stdout:**
```console
Traceback (most recent call last):
  File "/usr/bin/g910-gkeys", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/g910_gkeys/g910_gkeys.py", line 66, in main
    start()
  File "/usr/lib/python3.11/site-packages/g910_gkeys/g910_gkeys.py", line 93, in start
    device.disable_fkey_to_gkey_binding()
  File "/usr/lib/python3.11/site-packages/g910_gkeys/lib/usb_device.py", line 52, in disable_fkey_to_gkey_binding
    byte_written = self.dev.ctrl_transfer(0x21, 0x09, 0x0212, 1, packet, self.usb_timeout)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/core.py", line 1082, in ctrl_transfer
    ret = self._ctx.backend.ctrl_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 893, in ctrl_transfer
    ret = _check(self.lib.libusb_control_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
```

**g910-gkeys.log:**
```log
2023-09-23 18:22:23,896 [INFO] {g910_gkeys.g910_gkeys} - --------------------------------------------------------------------------------
2023-09-23 18:22:23,896 [INFO] {g910_gkeys.g910_gkeys} - ----------------------STARTED g910-keys-pid:64140------------------------------
2023-09-23 18:22:23,896 [INFO] {g910_gkeys.g910_gkeys} - --------------------------------------------------------------------------------
2023-09-23 18:22:23,910 [INFO] {g910_gkeys.lib.usb_device} - Keyboard Logitech G915 (wired) found.
2023-09-23 18:22:23,910 [DEBUG] {g910_gkeys.lib.usb_device} - usb dev:     INTERFACE 1: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x2
     iInterface         :    0x0 
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
2023-09-23 18:22:23,910 [DEBUG] {g910_gkeys.lib.usb_device} - Device default timeout: 1000ms
2023-09-23 18:22:23,910 [INFO] {g910_gkeys.misc.config} - Reading config from /home/jannik/.config/g910-gkeys/config.json
2023-09-23 18:22:24,411 [INFO] {g910_gkeys.lib.usb_device} - Trying to disable f-key to g-key binding
2023-09-23 18:22:24,411 [DEBUG] {g910_gkeys.lib.usb_device} - Sending ctrl command to keyboard b'\x11\x01\x11\x1a\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'...
```

I used the default `config.json` and tried both "de" and "en" settings.

suabo, 2023-09-23:
Normally the driver should install as a service and don't need to be run via console. This is indicating a problem with the installation. But if not run as a service it should also start via console. I think the problem may be user permissions on usb. There should be a file copied due to installation which you are probably missing. 
Do you have a file /etc/udev/rules.d/60-g910-gkeys.rules?

If not you can just create the file and copy paste the [content from the repo](https://github.com/JSubelj/g910-gkey-macro-support/blob/master/etc/60-g910-gkeys.rules).  
You will need root privileges for creating a file at that location!
After inserting the file you also need to restart udev service to initialize the new permissions:
`sudo systemctl restart udev`

After that you can try again. Can you also provide the output of `g910-gkeys -v`?

suabo, 2024-05-27:
I'll close this issue since there is no more response.
G915 support is integrated, but never been tested by myself.

closed 2024-05-27 by suabo