JSubelj/g910-gkey-macro-support #56: [Issue] Typeout `"\`"` doesn't work

Ginden, 2022-01-19:
**Describe the bug**

My config includes:

```
    "g1": {
        "hotkey_type": "typeout",
        "do": "`"
    },
```

When I press G1, it types out: `§ `.

If I change `G1` to any other string, eg. `foobar`, it works correctly.

**To Reproduce**

**Expected behavior**

I expected backtick to be typed out.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info**
 - did you run in daemon or normal mode
 - logs from /var/log/g910-gkeys.log
 - logs from jurnalctl -u g910-gkeys
 - configuration in /etc/g910-gkeys/config.json

**Desktop (please complete the following information):**
 - OS: Ubuntu 20
 - Kernel version
 - Gnome
 - g910-gkeys 0.2.4
 - python 2.7.18 or 3.8.10

**Additional context**
Add any other context about the problem here.

Ginden, 2022-01-19:
Responsible line:

https://github.com/JSubelj/g910-gkey-macro-support/blob/master/lib/data_mappers/char_uinput_mapper.py#L177

JSubelj, 2022-01-19:
Hey,

what keyboard do you have set up on your OS and what keyboard mapping do you have setup in your config? If those two don't match then you will have problems with different mapping. You can always write your own mapping for your language or you can use shortcut hotkey type and type out the wanted character.

Ginden, 2022-01-19:
@JSubelj Changing keyboard mapping to `en` fixed a issue.

Though, it's very surprising behaviour.

As side note - can you review #47 ?

closed 2022-01-19 by Ginden