JSubelj/g910-gkey-macro-support #62: [Issue]

Simon1511, 2022-10-29:
**Describe the bug**
So, as follows: I just updated to the latest version of this tool but couldn't get it to run. I had the config updated accordingly for the new version and the systemd service was running just fine (I didn't notice any errors). Out of frustration of the tool suddenly breaking I pressed my M1 button on the keyboard to load a config and that made the tool load the correct configs and my G-Keys started working.
I'm unsure whether this is an issue on my end or not or if this tool is just not properly loading the different profiles, or better said isn't loading a default profile.

**To Reproduce**
Steps to reproduce the behavior:
1. Install g910-gkeys
2. Adjust the config, for example add a typeout to G-Key 1 of profile 1
3. Notice that pressing the key will do nothing, even after reloading the systemd service.

**Expected behavior**
A clear and concise description of what you expected to happen.
A profile should be loaded by default instead of only loading one when M1/M2/M3 is pressed on the keyboard.

**Screenshots**
If applicable, add screenshots to help explain your problem.
None

**Additional info**
 - did you run in daemon or normal mode: Daemon (Systemd) mode
 - logs from /var/log/g910-gkeys.log: Attached to this post.
 - logs from jurnalctl -u g910-gkeys: See below.
 - configuration in /etc/g910-gkeys/config.json: Attached to this post.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Kernel version: 5.15.0-52-generic
 - desktop enviroment: Gnome 42.4
 - g910-gkeys version: 0.2.4
 - python version: 2.7.18 and 3.10.6 are installed.

**Additional context**
Add any other context about the problem here.

journalctl: (Don't wonder, I restarted the service very very often)
```
-- Boot fca965a33fca4797834d668f414b6484 --
Okt 29 19:11:50 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:29:26 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:29:26 simon-ubuntu env[1325]: Starting g910-gkeys, PID: 1325, logging at: /var/log/g910-gkeys.log
Okt 29 19:29:26 simon-ubuntu env[1325]: Got signal, SIGTERM terminating!
Okt 29 19:29:26 simon-ubuntu env[1325]: Removed uinput device
Okt 29 19:29:26 simon-ubuntu env[1325]: Exiting
Okt 29 19:29:26 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:29:26 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:29:26 simon-ubuntu systemd[1]: g910-gkeys.service: Consumed 13.414s CPU time.
Okt 29 19:29:26 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:30:08 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:30:08 simon-ubuntu env[14092]: Starting g910-gkeys, PID: 14092, logging at: /var/log/g910-gkeys.log
Okt 29 19:30:08 simon-ubuntu env[14092]: Got signal, SIGTERM terminating!
Okt 29 19:30:08 simon-ubuntu env[14092]: Removed uinput device
Okt 29 19:30:08 simon-ubuntu env[14092]: Exiting
Okt 29 19:30:08 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:30:08 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:30:08 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:32:30 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:32:30 simon-ubuntu env[14199]: Starting g910-gkeys, PID: 14199, logging at: /var/log/g910-gkeys.log
Okt 29 19:32:30 simon-ubuntu env[14199]: Got signal, SIGTERM terminating!
Okt 29 19:32:30 simon-ubuntu env[14199]: Removed uinput device
Okt 29 19:32:30 simon-ubuntu env[14199]: Exiting
Okt 29 19:32:30 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:32:30 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:32:30 simon-ubuntu systemd[1]: g910-gkeys.service: Consumed 1.722s CPU time.
-- Boot 0ed4c2c98e644acaa26674ce86f0f87a --
Okt 29 19:32:59 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:35:03 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:35:03 simon-ubuntu env[1328]: Starting g910-gkeys, PID: 1328, logging at: /var/log/g910-gkeys.log
Okt 29 19:35:03 simon-ubuntu env[1328]: Got signal, SIGTERM terminating!
Okt 29 19:35:03 simon-ubuntu env[1328]: Removed uinput device
Okt 29 19:35:03 simon-ubuntu env[1328]: Exiting
Okt 29 19:35:03 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:35:03 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:35:03 simon-ubuntu systemd[1]: g910-gkeys.service: Consumed 1.278s CPU time.
Okt 29 19:35:03 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:35:14 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:35:14 simon-ubuntu env[6875]: Starting g910-gkeys, PID: 6875, logging at: /var/log/g910-gkeys.log
Okt 29 19:35:14 simon-ubuntu env[6875]: Got signal, SIGTERM terminating!
Okt 29 19:35:14 simon-ubuntu env[6875]: Removed uinput device
Okt 29 19:35:14 simon-ubuntu env[6875]: Exiting
Okt 29 19:35:14 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:35:14 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:35:15 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:35:51 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:35:51 simon-ubuntu env[6922]: Starting g910-gkeys, PID: 6922, logging at: /var/log/g910-gkeys.log
Okt 29 19:35:51 simon-ubuntu env[6922]: Got signal, SIGTERM terminating!
Okt 29 19:35:51 simon-ubuntu env[6922]: Removed uinput device
Okt 29 19:35:51 simon-ubuntu env[6922]: Exiting
Okt 29 19:35:51 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:35:51 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:35:51 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:38:24 simon-ubuntu systemd[1]: Stopping Support for Logitech g910 keyboard gkeys...
Okt 29 19:38:24 simon-ubuntu env[7159]: Starting g910-gkeys, PID: 7159, logging at: /var/log/g910-gkeys.log
Okt 29 19:38:24 simon-ubuntu env[7159]: Got signal, SIGTERM terminating!
Okt 29 19:38:24 simon-ubuntu env[7159]: Removed uinput device
Okt 29 19:38:24 simon-ubuntu env[7159]: Exiting
Okt 29 19:38:24 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:38:24 simon-ubuntu systemd[1]: Stopped Support for Logitech g910 keyboard gkeys.
Okt 29 19:38:24 simon-ubuntu systemd[1]: g910-gkeys.service: Consumed 1.893s CPU time.
Okt 29 19:38:24 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:40:17 simon-ubuntu env[10693]: Starting g910-gkeys, PID: 10693, logging at: /var/log/g910-gkeys.log
Okt 29 19:40:17 simon-ubuntu env[10693]: Got signal, SIGTERM terminating!
Okt 29 19:40:17 simon-ubuntu env[10693]: Removed uinput device
Okt 29 19:40:17 simon-ubuntu env[10693]: Exiting
Okt 29 19:40:17 simon-ubuntu systemd[1]: g910-gkeys.service: Deactivated successfully.
Okt 29 19:40:17 simon-ubuntu systemd[1]: g910-gkeys.service: Consumed 1.291s CPU time.
Okt 29 19:40:30 simon-ubuntu systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Okt 29 19:41:05 simon-ubuntu su[11096]: (to simon) root on none
Okt 29 19:41:05 simon-ubuntu su[11096]: pam_unix(su:session): session opened for user simon(uid=1000) by (uid=0)
Okt 29 19:41:05 simon-ubuntu su[11096]: pam_unix(su:session): session closed for user simon
``

Logs and config:

[g910-gkeys.log](https://github.com/JSubelj/g910-gkey-macro-support/files/9894276/g910-gkeys.log)
[config.json.txt](https://github.com/JSubelj/g910-gkey-macro-support/files/9894281/config.json.txt)

suabo, 2022-10-31:
Sorry for that, i've already fixed this issue in my current [dev branch](https://github.com/suabo/g910-gkey-macro-support/tree/dev-0.2.x)
Pull request for it is coming soon.

Simon1511, 2022-10-31:
> Sorry for that, i've already fixed this issue in my current [dev branch](https://github.com/suabo/g910-gkey-macro-support/tree/dev-0.2.x) Pull request for it is coming soon.

Ah okay, thank you very much!

closed 2022-10-31 by Simon1511