JSubelj/g910-gkey-macro-support #76: [Issue] StopIteration causes service to not start

Spolonus282, 2023-02-02:
**Describe the bug**
Upon starting the service, either through computer startup or manually, nothing occurs. The logs state that a StopIteration occurred.

**To Reproduce**
Steps to reproduce the behavior:
0. have g910-gkeys installed
1. run 'g910-gkeys'
2. See error

**Expected behavior**
The service should start without errors

**Screenshots**
[image]

**Additional info**
 - run in normal mode
 - no relevant logs apart from above (see additional context for why)
 - config.json: 
 ```json
{
    "__comment": "following hotkey types are supported: nothing, typeout, shortcut and run; only en and si keyboard mappings are currently supported",
    "keyboard_mapping": "en",
    "notify": "False",
    "username": "",
    "profiles": {
        "m1": {
            "g1": {
                "hotkey_type": "typeout",
                "do": "test"
            },
            "g2": {
                "hotkey_type": "nothing",
                "do": ""
            },
            "g3": {
                "hotkey_type": "shortcut",
                "do": "ctrl+v"
            },
            "g4": {
                "hotkey_type": "shortcut",
                "do": "ctrl+j"
            },
            "g5": {
                "hotkey_type": "shortcut",
                "do": "alt+F1"
            },
            "g6": {
                "hotkey_type": "nothing",
                "do": ""
            },
            "g7": {
                "hotkey_type": "run",
                "do": "su <username> -c 'gnome-screenshot -ac' & 2>&1> > /dev/null"
            },
            "g8": {
                "hotkey_type": "shortcut",
                "do": "ctrl+pagedown"
            },
            "g9": {
                "hotkey_type": "shortcut",
                "do": "ctrl+pageup"
            }
        }
    }
}
 ```

**Desktop:**
 - OS: Linux Mint 21.1
 - Kernel version: 5.15.0-58-generic
 - desktop environment: Cinnamon 5.6.7
 - g910-gkeys version: 0.2.5
 - python version: 3.10.6

**Additional context**
The problem occurred directly after updating from 0.2.4 to 0.2.5. Version 0.2.4 was working fine.

suabo, 2023-02-02:
Did you use the installer.sh to update the driver?

Can you try to start it as a service `sudo systemctl start g910-gkeys`?

Spolonus282, 2023-02-02:
Yes, I did use installer.sh to update after manually uninstalling and pulling per the upgrade wiki page.
After running the above command I got the same error as before.

suabo, 2023-02-02:
You normally don't have to run `g910-gkeys` as you described, since it is designed to run as a service.

Can you please do the following:
1. Make sure the service is stopped `sudo systemctl stop g910-gkeys`
2. Delete the log file `/var/log/g910-gkeys.log`
3. Start the service `sudo systemctl start g910-gkeys`

Now the driver should be working. Try to hit macro key **G1**
Take a look at the log file if it is still not working and post the content here.

If you still got the same error i assume the installation went wrong and you maybe have to try the manual installation with `sudo python3 setup.py install` to see if there is any errors. You will find more information about manual install [here](https://github.com/JSubelj/g910-gkey-macro-support/wiki/Detailed-installation#build-package-from-source).

Spolonus282, 2023-02-03:
I stopped the service, deleted the log, started the service and it still didn't work
I uninstalled it using `sudo ./uninstall.sh` then reinstalled using the above command. After re-enabling the service and starting it, the same error occurred.

Spolonus282, 2023-02-04:
Sorry, I didn't see the request to see the log file. After deleting it as you requested, no new log file has been generated. I suspect the service is closing immediately, before it has a chance to run anything.

Spolonus282, 2023-02-08:
I have solved the issue. It seems that the old 0.2.4 package folder was not removed on uninstall for some reason. All files were removed, leaving only empty directories, meaning pip couldn't identify it and allow me to uninstall it.
However, the main service module still considered it the default location (?) to check for the files, of which there were none, causing the above error.
Removing the directory from dist-packages and starting the service got it to work again.
I appreciate the help and am closing the issue.

closed 2023-02-08 by Spolonus282