JSubelj/g910-gkey-macro-support #78: [Feature] Allow looping of configured action

rastilion, 2023-05-21:
It would be nice to have a toggle for the config to allow looping a configured action.
For example configuring typeout to write "hello " as long as configured key is pressed, resulting  in a chain of "hello hello hello ... ", or using "shortcut" configured with "Ctrl+W" to close browser tabs while the key is held. 
An advanced config could also feature a delay between loops in ms.
Example config for this scenario:
```
"g1": {
    "hotkey_type": "typeout",
    "do": "hello ",
    "loop": true,
    "delay": 500
}
```
would type "hello " every 500ms while the g1-key is held.

Thank you for this great tool btw, i have been looking for quite a while!

suabo, 2023-09-10:
@rastilion Thank you for this nice idea!

Since it is the same with the normal keys the behavior will be as expected by the user.

Maybe there is no need to set the delay for every single macro key? I would start with an global delay setting for all macro keys. Maybe you can proof me wrong!

rastilion, 2023-09-30:
@suabo The reason I suggested the per-key delay comes from a gaming background.
In most mmos you can set up keybinds with abilities, so a single key can trigger multiple abilities.
Being able to stagger the execution there or taking ability cooldowns into consideration when setting up those keys can really help in those instances.

A global delay might be useful as well, but i'm failing to see a scenario here.