JSubelj/g910-gkey-macro-support #79: [Issue]

wgcj, 2023-08-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Nearly installed Ubuntu Lunar 23.04
2. Ran these commands (grabbed from my history)
   69  modprobe uinput
   70  git clone https://github.com/JSubelj/g910-gkey-macro-support.git
   71  sudo apt install git
   72  git clone https://github.com/JSubelj/g910-gkey-macro-support.git
   73  cd g910-gkey-macro-support
   74  chmod +x installer.sh; sudo ./installer.sh
  See attachment "installer-output.txt"
  Started going through the troubleshooting and saw that I had python 3.11 as I did these commands
  76  sudo apt-get install python3-setuptools
   77  sudo ./installer.sh
   78  pip install -U pip setuptools 
   79  sudo apt install python3-pip
   80  sudo ./installer.sh
   81  sudo python3 setup.py install --record files.txt
   82  sudo g910-gkeys -v
   83  python3 --version
   84  python3.7 --version
   85  sudo python3.7 setup.py install

whitney@valentina:~/Documents/Installations/g910-gkey-macro-support$ python3 --version
Python 3.11.4
whitney@valentina:~/Documents/Installations/g910-gkey-macro-support$ 

Started trying to install 3.7, but looks like deadsnakes ppa never made it for 23.04, and that 3.7's eol was a few months ago. 

**Expected behavior**
I want to know if the installation is requiring specifically 3.7 or just 3.7 or later. If it's specifically requiring 3.7, it looks like I need to stop while I'm ahead and just wait to see if your project is interested in moving to a non-eol python

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info**
 - did you run in daemon or normal mode: Normal
 - logs from /var/log/g910-gkeys.log: n/a as installation didn't complete
 - logs from jurnalctl -u g910-gkeys: n/a as installation didn't complete
 - configuration in /etc/g910-gkeys/config.json: n/a as installation didn't complete

**Desktop (please complete the following information):**
 - OS: Ubuntu 23.04
 - Kernel version: 6.2.0-27-generic
 - desktop enviroment: gnome
 - g910-gkeys version: n/a as haven't installed, but just trying to install the latest from github
 - python version: 3.11

**Additional context**
It seems from your documentation that it's only 3.7, but I wanted to verify before I gave up. I've used this for years (new build) and I've really appreciated all the additional flexibility from the extra buttons that, ultimately, this project gives me and not logitech.

suabo, 2023-09-06:
Where should I look for the attachment "installer-output.txt".
I can see no error message in your description.

suabo, 2023-09-07:
I checked this issue again and installed a few versions of python3 to double check that issue.
The only version I could get it to work with was python3.8 everything above showed some different errors with the dependencies.
Can you try to install 3.8 and test it again with the v0.3.0?

The oel of 3.8 is in about an year from now.

suabo, 2023-12-05:
Support for python 3.11 was added with version 0.4.0 so I'll close this issue.

closed 2023-12-05 by suabo