JSubelj/g910-gkey-macro-support #80: [Issue] Line ending with \n is not working with keymapping de

Zalkortis, 2023-08-28:
**Describe the bug**
When I setup a GKey to typeout "\n/remaining\n", I get the following error:

2023-08-28 19:28:48,138 [WARNING] {lib.data_mappers.config_reader} - Error(s) in config: {'m1': {'g5': {'do': {'typeout': 'Character \n on position 10 for keyboard mapping: de does not exist! '}}}}

When I change the keymapping to en it works, but then the "/"-character is identified as "-"-character. I didn't found out which character I need on the German keyboard to let the application identify it as "/"-character.

**To Reproduce**
Steps to reproduce the behavior:
1. Open configuration
2. change keyboard_mapping to "de"
3. Set a gkey to the following action:
      "g5": {
        "hotkey_type": "typeout",
        "do": "\n/remaining\n"
        }
4. See error in logfile

**Expected behavior**
It should work the same way as with keymapping in en.

**Screenshots**
https://imgur.com/a/otdStN4

**Additional info**
 - did you run in daemon or normal mode -> both
 - logs from /var/log/g910-gkeys.log  -> See screenshot above
 - logs from journalctl -u g910-gkeys  -> See screenshot above
 - configuration in /etc/g910-gkeys/config.json  --> https://pastebin.com/7FMXwtP0

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Kernel version 6.4.12-zen1-1-zen
 - desktop enviroment KDE Plasma 5.27.7
 - g910-gkeys version g910-gkeys-git 0.2.5.r0.g600610b-1
 - python version Python 3.11.3

**Additional context**
Add any other context about the problem here.

Zalkortis, 2023-08-28:
I've also just checked the code, it seems the 'de' object is incomplete. at least these definitions are missing:
        '\t': click(uinput.KEY_TAB),
        '\n': click(uinput.KEY_ENTER),
        ' ': click(uinput.KEY_SPACE),

I've added these lines to the local file 
/usr/lib/python3.11/site-packages/lib/data_mappers/char_uinput_mapper.py

restarted the service, and it works.

Also when I compare the language definitions, it seems there is missing a lot and it doesn't look consistent. Please have a look and check to have it all the same.

suabo, 2023-09-06:
In the new version 0.3.0 the keys are now included.
The driver has many other improvements. So maybe you want to update.

closed 2023-09-06 by suabo