JSubelj/g910-gkey-macro-support #83: [Issue] September 17th 2023 update broke the service

Captn138, 2023-09-18:
**Describe the bug**
App was working before. Update released on september 17th. App is now broken

**To Reproduce**
Steps to reproduce the behavior:
Download and run the service.

**Expected behavior**
Service should not crash

**Screenshots**
N/A

**Additional info**
 - did you run in daemon or normal mode
systemd daemon
 - logs from /var/log/g910-gkeys.log
no more logs since update
 - logs from jurnalctl -u g910-gkeys
```Sep 18 18:28:31 z9-neo systemd[1]: g910-gkeys.service: Scheduled restart job, restart counter is at 22.
Sep 18 18:28:31 z9-neo systemd[1]: Started Support for Logitech g910 keyboard gkeys.
Sep 18 18:28:31 z9-neo env[46431]: Traceback (most recent call last):
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/bin/g910-gkeys", line 33, in <module>
Sep 18 18:28:31 z9-neo env[46431]:     sys.exit(load_entry_point('g910-gkeys==0.4.0', 'console_scripts', 'g910-gkeys')())
Sep 18 18:28:31 z9-neo env[46431]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/bin/g910-gkeys", line 25, in importlib_load_entry_point
Sep 18 18:28:31 z9-neo env[46431]:     return next(matches).load()
Sep 18 18:28:31 z9-neo env[46431]:            ^^^^^^^^^^^^^^^^^^^^
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
Sep 18 18:28:31 z9-neo env[46431]:     module = import_module(match.group('module'))
Sep 18 18:28:31 z9-neo env[46431]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
Sep 18 18:28:31 z9-neo env[46431]:     return _bootstrap._gcd_import(name[level:], package, level)
Sep 18 18:28:31 z9-neo env[46431]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Sep 18 18:28:31 z9-neo env[46431]:   File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
Sep 18 18:28:31 z9-neo env[46431]:   File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
Sep 18 18:28:31 z9-neo env[46431]:   File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
Sep 18 18:28:31 z9-neo env[46431]:   File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
Sep 18 18:28:31 z9-neo env[46431]:   File "<frozen importlib._bootstrap_external>", line 940, in exec_module
Sep 18 18:28:31 z9-neo env[46431]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/lib/python3.11/site-packages/g910_gkeys/g910_gkeys.py", line 9, in <module>
Sep 18 18:28:31 z9-neo env[46431]:     from g910_gkeys.misc.config import Config
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/lib/python3.11/site-packages/g910_gkeys/misc/config.py", line 14, in <module>
Sep 18 18:28:31 z9-neo env[46431]:     class Config:
Sep 18 18:28:31 z9-neo env[46431]:   File "/usr/lib/python3.11/site-packages/g910_gkeys/misc/config.py", line 20, in Config
Sep 18 18:28:31 z9-neo env[46431]:     config_dir: str = os.getenv("HOME") + "/.config/g910-gkeys"
Sep 18 18:28:31 z9-neo env[46431]:                       ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
Sep 18 18:28:31 z9-neo env[46431]: TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
Sep 18 18:28:31 z9-neo systemd[1]: g910-gkeys.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 18:28:31 z9-neo systemd[1]: g910-gkeys.service: Failed with result 'exit-code'.
```
 - configuration in /etc/g910-gkeys/config.json
```
{
  "__comment" : "following hotkey types are supported: nothing, typeout, shortcut and run; only en, fr, de and si keyboard mappings are currently supported",
  "keyboard_mapping": "fr",
  "notify": "True",
  "username": "captn",
  "profiles": {
    "MEMORY_1": {
      "MACRO_1": {
        "hotkey_type": "uinput",
        "do": "KEY_F13"
      },
      "MACRO_2": {
        "hotkey_type": "uinput",
        "do": "KEY_F14"
      },
      "MACRO_3": {
        "hotkey_type": "uinput",
        "do": "KEY_F15"
      },
      "MACRO_4": {
        "hotkey_type": "uinput",
        "do": "KEY_F16"
      },
      "MACRO_5": {
        "hotkey_type": "uinput",
        "do": "KEY_F17"
      },
      "MACRO_6": {
        "hotkey_type": "uinput",
        "do": "KEY_F18"
      },
      "MACRO_7": {
        "hotkey_type": "uinput",
        "do": "KEY_F19"
      },
      "MACRO_8": {
        "hotkey_type": "uinput",
        "do": "KEY_F20"
      },
      "MACRO_9": {
        "hotkey_type": "uinput",
        "do": "KEY_F21"
      }
    },
    "MEMORY_2": {
      "MACRO_1": {
        "hotkey_type": "uinput",
        "do": "KEY_F13"
      },
      "MACRO_2": {
        "hotkey_type": "uinput",
        "do": "KEY_F14"
      },
      "MACRO_3": {
        "hotkey_type": "uinput",
        "do": "KEY_F15"
      },
      "MACRO_4": {
        "hotkey_type": "uinput",
        "do": "KEY_F16"
      },
      "MACRO_5": {
        "hotkey_type": "uinput",
        "do": "KEY_F17"
      },
      "MACRO_6": {
        "hotkey_type": "uinput",
        "do": "KEY_F18"
      },
      "MACRO_7": {
        "hotkey_type": "uinput",
        "do": "KEY_F19"
      },
      "MACRO_8": {
        "hotkey_type": "uinput",
        "do": "KEY_F20"
      },
      "MACRO_9": {
        "hotkey_type": "uinput",
        "do": "KEY_F21"
      }
    },
    "MEMORY_3": {
      "MACRO_1": {
        "hotkey_type": "uinput",
        "do": "KEY_F13"
      },
      "MACRO_2": {
        "hotkey_type": "uinput",
        "do": "KEY_F14"
      },
      "MACRO_3": {
        "hotkey_type": "uinput",
        "do": "KEY_F15"
      },
      "MACRO_4": {
        "hotkey_type": "uinput",
        "do": "KEY_F16"
      },
      "MACRO_5": {
        "hotkey_type": "uinput",
        "do": "KEY_F17"
      },
      "MACRO_6": {
        "hotkey_type": "uinput",
        "do": "KEY_F18"
      },
      "MACRO_7": {
        "hotkey_type": "uinput",
        "do": "KEY_F19"
      },
      "MACRO_8": {
        "hotkey_type": "uinput",
        "do": "KEY_F20"
      },
      "MACRO_9": {
        "hotkey_type": "uinput",
        "do": "KEY_F21"
      }
    }
  }
}
```

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Kernel version: 6.5.3-zen1-1-zen
 - desktop enviroment: Gnome 44 on Wayland
 - g910-gkeys version: 0.3.0.r16.g43fafc8-1
 - python version: 3.11.5

**Additional context**
afaik the issue would come from reading the $HOME env var and returning NoneType object instead of a string.

suabo, 2023-09-18:
Can you help me figure out why it is not working for you?
Can you first try `echo $HOME`. The response should be _/home/yourusername_

Since the application runs no longer with root privileges it needs to know your home folder to read/write config and logs.
The log file should now be in your home directory. The config file should be in `~/.config/g910-gkeys/config.json`. The installer should have copied your existing config to that location. Or maybe not if the HOME env var is not set?

suabo, 2023-09-18:
Since the service is now installed as user service you have to use:
journalctl --user -u g910-gkeys

The fact you get a message in `journalctl -u g910-gkeys` is dedicating the service is still run by root.

Can you confirm you didn't install with root (sudo)?
If not try to uninstall and install **without sudo** again.  
That part changed in the installation since the last version!

Captn138, 2023-09-18:
Well I think there is the broblem.
The service still runs as root, and .config/g910-gkeys doesn't exist.
I installed it using yay on ArchLinux.
```
$ systemctl cat g910-gkeys

# /usr/lib/systemd/system/g910-gkeys.service
[Unit]
Description=Support for Logitech g910 keyboard gkeys
Documentation=https://github.com/JSubelj/g910-gkey-macro-support/wiki

[Service]
Type=simple
ExecStart=/usr/bin/env g910-gkeys
ExecReload=/bin/kill -SIGUSR1 $MAINPID
TimeoutStopSec=5
Restart=on-failure
RestartSec=10

[Install]
WantedBy=default.target
```

suabo, 2023-09-19:
I forgot about the PKGBUILD. It still sets the service up for root.
I try to correct it, but since I'm in ubuntu I can't really test it.

It seams like we also got an issue with versioning in arch distributions because the git path to the repo in PKGBUILD points to the master branch, which is always the latest version. Meaning that you can't even use the old 0.3.0 with yay, because PKGBUILD will get the latest version. This is just an hint for me to fix this issue, too.

Captn138, 2023-09-21:
Okay, thanks. I'll probably wait for the AUR update tho.

Zalkortis, 2023-09-24:
I've downloaded v0.3.0 as source, removed the AUR package, because non was working anymore. Installed 0.3.0 manually and now its working again. I think I'll just stick with 0.3.0 now, don't see a point in spending that much time every once in a while to try to fix the issues with this tool.

I would highly recommend to test your stuff before breaking it for everybody and then don't come back to fix it.

suabo, 2023-09-24:
I'm sorry for the inconvenience. At the moment there is nobody testing the driver on arch linux and since the last update brought a big change the chances something breaking was very high. I just set up an arch linux in docker and trying my best to fix the aur package. I report here if there are some news.

jlaunay, 2023-09-26:
The [package](https://aur.archlinux.org/packages/g910-gkeys-git) was updated on AUR on 09/22 and it works (Feel free to leave a comment on AUR if it's not the case).
Do not use the PKGBUILD available here, packages should never handle your /home directory (pacman is running as root).
The PKGBUILD [here](https://github.com/JSubelj/g910-gkey-macro-support/blob/master/pkgbuild/PKGBUILD) cannot work 

- `pip install` is for user, on arch you should use `python -m installer` in a PKGBUILD.
- In the post_install script, $HOME will always point to user "root".
- A systemd user service have to be install in `/usr/lib/systemd/user` and not in `$HOME"/.config/systemd/user`

Captn138, 2023-09-27:
I think something is wrong then, because it says there is a permission error:

```
sept. 27 21:24:49 z9-neo env[28876]: Traceback (most recent call last):
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/bin/g910-gkeys", line 8, in <module>
sept. 27 21:24:49 z9-neo env[28876]:     sys.exit(main())
sept. 27 21:24:49 z9-neo env[28876]:              ^^^^^^
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/g910_gkeys/g910_gkeys.py", line 66, in main
sept. 27 21:24:49 z9-neo env[28876]:     start()
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/g910_gkeys/g910_gkeys.py", line 83, in start
sept. 27 21:24:49 z9-neo env[28876]:     device = USBDevice()
sept. 27 21:24:49 z9-neo env[28876]:              ^^^^^^^^^^^
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/g910_gkeys/lib/usb_device.py", line 22, in __init__
sept. 27 21:24:49 z9-neo env[28876]:     if self.dev.is_kernel_driver_active(self.usb_if) is True:
sept. 27 21:24:49 z9-neo env[28876]:        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/usb/core.py", line 1107, in is_kernel_driver_active
sept. 27 21:24:49 z9-neo env[28876]:     self._ctx.managed_open()
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/usb/core.py", line 113, in wrapper
sept. 27 21:24:49 z9-neo env[28876]:     return f(self, *args, **kwargs)
sept. 27 21:24:49 z9-neo env[28876]:            ^^^^^^^^^^^^^^^^^^^^^^^^
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/usb/core.py", line 131, in managed_open
sept. 27 21:24:49 z9-neo env[28876]:     self.handle = self.backend.open_device(self.dev)
sept. 27 21:24:49 z9-neo env[28876]:                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 804, in open_device
sept. 27 21:24:49 z9-neo env[28876]:     return _DeviceHandle(dev)
sept. 27 21:24:49 z9-neo env[28876]:            ^^^^^^^^^^^^^^^^^^
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 652, in __init__
sept. 27 21:24:49 z9-neo env[28876]:     _check(_lib.libusb_open(self.devid, byref(self.handle)))
sept. 27 21:24:49 z9-neo env[28876]:   File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 604, in _check
sept. 27 21:24:49 z9-neo env[28876]:     raise USBError(_strerror(ret), ret, _libusb_errno[ret])
sept. 27 21:24:49 z9-neo env[28876]: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
sept. 27 21:24:49 z9-neo systemd[1081]: g910-gkeys.service: Main process exited, code=exited, status=1/FAILURE
```

suabo, 2023-09-27:
Thx @jlaunay for the correction of the pkgbuild. I will merge it into the repo as well.
Maybe you forgot to tell people to reload the daemon, which is handling udev and the permissions for usb.
After installation you need to do `systemctl --user daemon-reload`
After reboot it should also work.

jlaunay, 2023-09-28:
@Captn138 The [package](https://aur.archlinux.org/packages/g910-gkeys-git) was updated on AUR, you shouldn't have any more permission problems.

@suabo

`60-g910-gkeys.rules` is not in the right place, in PKGBUILD you have `install -Dm 644 -t "$pkgdir"/etc/udev/rules.d/60-g910-gkeys.rules etc/60-g910-gkeys.rules` which result in 
`/etc/udev/rules.d/60-g910-gkeys.rules/60-g910-gkeys.rules` instead of
`/etc/udev/rules.d/60-g910-gkeys.rules`

You must correct with `install -Dm 644 -t $pkgdir/etc/udev/rules.d/ etc/60-g910-gkeys.rules`

Same thing with `uinput-g910-gkeys.conf`

`systemctl --user daemon-reload` does not reload udev rules, it reload systemd manager configuration. This will rerun all generators, reload all unit files, and recreate the entire dependency tree. You need to run this command if your .service file changed on disk.

udev and the permissions for usb are handling by udev and to reload rules without reboot you can run as root

`udevadm control --reload-rules &&  udevadm trigger`

Everything is on AUR.

suabo, 2023-09-28:
Thx @jlaunay I fixed the PKGBUILD and .install file as you mentioned.
Hope it works again for everybody now.

closed 2024-01-06 by Captn138