JSubelj/g910-gkey-macro-support #84: [Issue] scripts requiring root / sudo privileges no longer work

Ultrabenosaurus, 2023-10-02:
**Describe the bug**
I have a couple of Gkeys mapped to scripts which require root privileges to run, for example switching wireguard profiles or mounting / unmounting NAS shares.

I had some trouble with the `g910-gkeys` service recently and nothing was working following an update, but thankfully after another update it started working again.

However, now these scripts will not run and also have the knock-on effect of locking my account for 10 minutes! Very frustrating!

Is there a way of having g910-gkeys run scripts that require root again?

**To Reproduce**
Steps to reproduce the behavior:
1. have a script or command that requires root / sudo privileges
2. map it to a gkey
3. press the gkey

**Expected behavior**
The script runs successfully as it used to.

**Screenshots**
N/A - see output of `journalctl --user --user-unit=g910-gkeys` below.

**Additional info**

```
Oct 02 14:09:11 Manjaro systemd[1711]: Started Support for Logitech g910 keyboard gkeys.
Oct 02 21:38:43 Manjaro sudo[422381]: pam_unix(sudo:auth): conversation failed
Oct 02 21:38:43 Manjaro sudo[422381]: pam_unix(sudo:auth): auth could not identify password for [dan]
Oct 02 21:38:46 Manjaro sudo[422381]: PAM Attempted to close sd-bus after fork, this should not happen.
Oct 02 21:38:46 Manjaro sudo[422443]: pam_unix(sudo:auth): conversation failed
Oct 02 21:38:46 Manjaro sudo[422443]: pam_unix(sudo:auth): auth could not identify password for [dan]
Oct 02 21:38:48 Manjaro sudo[422443]: PAM Attempted to close sd-bus after fork, this should not happen.
Oct 02 21:39:03 Manjaro sudo[422752]: pam_unix(sudo:auth): conversation failed
Oct 02 21:39:03 Manjaro sudo[422752]: pam_unix(sudo:auth): auth could not identify password for [dan]
Oct 02 21:39:03 Manjaro sudo[422752]: pam_faillock(sudo:auth): Consecutive login failures for user dan account temporarily locked
Oct 02 21:39:05 Manjaro sudo[422752]: PAM Attempted to close sd-bus after fork, this should not happen.
Oct 02 21:39:05 Manjaro sudo[422815]: pam_unix(sudo:auth): conversation failed
Oct 02 21:39:05 Manjaro sudo[422815]: pam_unix(sudo:auth): auth could not identify password for [dan]
Oct 02 21:39:07 Manjaro sudo[422815]: PAM Attempted to close sd-bus after fork, this should not happen.
Oct 02 21:52:50 Manjaro sudo[436264]: pam_unix(sudo:auth): conversation failed
```

**Desktop (please complete the following information):**
 - OS: Manjaro x86_64
 - Kernel: 6.5.3-1
 - DE: Plasma 5.27.8
 - g910-gkeys version: 0.4.0
 - python version: 3.11.5

**Additional context**
My scripts do not call `sudo`, but imagine the behaviour would be the same.

suabo, 2023-12-04:
Yes, that's true. Since the driver is now running with user privileges it's no longer possible to run scripts as root.
This was a security issue.

You need to manage some permissions so you don't need root for your scripts or go back to 0.3.0.
Saidly I think the version history on AUR is not working properly. So you should install from source.

To install 0.3.0 from source you maybe need python 3.8.

Please tell me if this helps?

Also if there is anybody else reading this with the same issue please make a comment so i know there is need to do something about it. Maybe I can make a 0.3.1 which is working with AUR.

suabo, 2024-05-27:
Since there is no more response on this topic I close this issue.

closed 2024-05-27 by suabo

braoult, 2024-05-28:
Anyway, I believe the correct way in this specific case was to use `sudo`, and configure it not to ask for a password for these commands,  something like in `/etc/sudoers`:
```
myuser myhost = (root) NOPASSWD: /usr/local/bin/my-command-as-root
```
I don't see why a single shortcut would be able to do something that yourself could not run in a terminal.

Another option is to use the setuid bit, but it could lead to much more security issues...