JSubelj/g910-gkey-macro-support #85: [Issue]  Service not found on Fedora 39

joelorentz, 2023-10-25:
The installer runs without issue for me but when i try to start the service i get :+1: 
`Failed to enable unit: Unit file g910-gkeys.service does not exist.
`

I am on Fedora 39, anyone has an idea?

suabo, 2023-12-04:
I learned that the paths where the unit files are stored differs between distros. The installer puts the file in `"$HOME"/.config/systemd/user/g910-gkeys.service` but on Fedora it needs to be in a different location. Unit files for user units are stored similarly in /etc/systemd/user, /run/systemd/user, and /usr/lib/systemd/user.

You need to copy the unit file to one of those locations and then enable and start the service:
`systemctl --user enable --now g910-gkeys.service`

bschatz, 2023-12-29:
In the instalöler script the command

``mkdir -p $HOME/.config/systemd/user/``

is missing.

execute it manually or add it to the script and run the installer.sh again.