JSubelj/g910-gkey-macro-support #86: [Issue] error: externally-managed-environment, service cant find g910-gkeys

bschatz, 2023-12-29:
**Describe the bug**
Installer script prints out error:

```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
```

The error comes from the line
``pip install -e ./ `` in the installer.sh script.

Journalctl shows error
``env[35229]: /usr/bin/env: „g910-gkeys“: Datei oder Verzeichnis nicht gefunden``

--> File not found.

**To Reproduce**
run ./installer.sh

**Expected behavior**
no error, g910-gkeys is found by service

**Screenshots**

**Additional info**
 - did you run in daemon or normal mode

as a systemd user service

 - logs from /var/log/g910-gkeys.log

does not exist

 - logs from journalctl -u g910-gkeys

```
$ journalctl --user -u g910-gkeys | tail -10
Dec 29 17:17:46 jupiter systemd[2940]: Started g910-gkeys.service - Support for Logitech g910 keyboard gkeys.
Dec 29 17:17:46 jupiter env[36110]: /usr/bin/env: „g910-gkeys“: Datei oder Verzeichnis nicht gefunden
Dec 29 17:17:46 jupiter systemd[2940]: g910-gkeys.service: Main process exited, code=exited, status=127/n/a
Dec 29 17:17:46 jupiter systemd[2940]: g910-gkeys.service: Failed with result 'exit-code'.
Dec 29 17:17:56 jupiter systemd[2940]: g910-gkeys.service: Scheduled restart job, restart counter is at 186.
Dec 29 17:17:56 jupiter systemd[2940]: Stopped g910-gkeys.service - Support for Logitech g910 keyboard gkeys.
Dec 29 17:17:56 jupiter systemd[2940]: Started g910-gkeys.service - Support for Logitech g910 keyboard gkeys.
Dec 29 17:17:56 jupiter env[36119]: /usr/bin/env: „g910-gkeys“: Datei oder Verzeichnis nicht gefunden
Dec 29 17:17:56 jupiter systemd[2940]: g910-gkeys.service: Main process exited, code=exited, status=127/n/a
Dec 29 17:17:56 jupiter systemd[2940]: g910-gkeys.service: Failed with result 'exit-code'.
```

 - configuration in /etc/g910-gkeys/config.json

Does not exist. Use origin config.json, located in 
```~/.config/g910-gkeys/config.json```

**Desktop (please complete the following information):**
 - OS: [e.g. manjaro]

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

 - Kernel version

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

 - desktop enviroment 
 
```
$ sudo apt show xfce4
Package: xfce4
Version: 4.18
```

 - g910-gkeys version

```
$ git describe --abbrev   
0.3.0-19-g4b1f05e
```

 - python version
```
$ apt show python3
Package: python3
Version: 3.11.2-1+b1
```

**Additional context**
Add any other context about the problem here.

suabo, 2024-05-27:
Can you ensure you have installed pyusb & python-uinput. You can use `pip list|grep py`.
You should see something like this:
```
...
python-uinput         0.11.2
...
pyusb                 1.2.1
...
```

If you don't see it here try to install it:
`sudo apt install python3-uinput python-usb`

After installing the required dependencies you should be able to run the install script with no errors.
Please also use the newest version!

BLACK4585, 2025-03-02:
I just go the same issue. I have the required Python libraries installed. What can I do?

schoenid, 2025-05-20:
Got the same errors.

After trying to install the missing modules:
`python-usb` is not available.
The package is called: `python3-usb`

Anyway, it still does not work.
`/usr/bin/env: »g910-gkeys“: Datei oder Verzeichnis nicht gefunden`

Where is this command located ?
(Or where should it be ?)

There is also another problem since I've installed the python3-usb package:
```
sd@Radieschen:~/Tech/Linux/g910-gkey-macro-support$ chmod +x install; ./installerror: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Created symlink /home/sd/.config/systemd/user/default.target.wants/g910-gkeys.service → /usr/lib/systemd/user/g910-gkeys.service.
```

Oh maybe the problem is: there is Python 3.12 installed.
How can I solve this?

```
~/Tech/Linux/g910-gkey-macro-support$ git describe --abbrev
0.4.2
```

OS: 
Operating System: Ubuntu Studio 24.04
KDE Plasma Version: 5.27.12
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.13
Kernel Version: 6.8.0-60-lowlatency (64-bit)
Graphics Platform: X11
Processors: 16 × 13th Gen Intel® Core™ i7-13700K
Memory: 62.5 GiB of RAM
Graphics Processor: NV196
Manufacturer: ASUS