JSubelj/g910-gkey-macro-support #87: [Issue] Media keys stopped working

Captn138, 2024-01-06:
**Describe the bug**
The STOP, NEXT and PREVIOUS media keys do not work while using the g910-gkeys driver.

**To Reproduce**
Steps to reproduce the behavior:
1. systemctl --user start g910-gkeys
3. I was messing with my keyboard shortcuts in Gnome, so in Settings, Keyboard, View and customize shortcuts
4. Try to remap a function with the aformentioned keys
5. Doesn't work
6. systemctl --user stop g910-gkeys
7. Test again
8. Works

**Expected behavior**
The keys should work as they do without the driver.

**Additional info**
 - I also tested on some media players like Spotify, Celluloid and Lollipop.
 - Althought I explained how I saw that bug, I do not know for how long this has been around, since I never used the media keys apart from PLAY/PAUSE.

**Desktop:**
 - OS: `Arch Linux`
 - Kernel version `6.6.9-zen1-1-zen`
 - desktop enviroment `Gnome 45.2` on `Wayland`
 - g910-gkeys version `0.4.0 (AUR)`
 - python version `3.11.6`
 - systemd service:
```
[Unit]
Description=Support for Logitech g910 keyboard gkeys
Documentation=https://github.com/JSubelj/g910-gkey-macro-support/wiki

[Service]
Type=simple
ExecStart=/usr/bin/env g910-gkeys
ExecReload=/bin/kill -SIGUSR1 $MAINPID
TimeoutStopSec=5
Restart=on-failure
RestartSec=10

[Install]
WantedBy=default.target
```
- config file:
```
{
  "__comment" : "following hotkey types are supported: nothing, typeout, shortcut and run; only en, fr, de and si keyboard mappings are currently supported",
  "keyboard_mapping": "fr",
  "notify": "False",
  "username": "captn",
  "profiles": {
    "MEMORY_1": {
      "MACRO_1": {
        "hotkey_type": "uinput",
        "do": "KEY_F14"
      },
      "MACRO_2": {
        "hotkey_type": "uinput",
        "do": "KEY_F15"
      },
      "MACRO_3": {
        "hotkey_type": "uinput",
        "do": "KEY_F16"
      },
      "MACRO_4": {
        "hotkey_type": "uinput",
        "do": "KEY_F17"
      },
      "MACRO_5": {
        "hotkey_type": "uinput",
        "do": "KEY_F18"
      },
      "MACRO_6": {
        "hotkey_type": "shortcut",
        "do": "alt+F1"
      },
      "MACRO_7": {
        "hotkey_type": "shortcut",
        "do": "alt+F2"
      },
      "MACRO_8": {
        "hotkey_type": "shortcut",
        "do": "alt+F3"
      },
      "MACRO_9": {
        "hotkey_type": "shortcut",
        "do": "alt+F12"
      }
    },
    "MEMORY_2": {
      "MACRO_1": {
        "hotkey_type": "uinput",
        "do": "KEY_M"
      },
      "MACRO_2": {
        "hotkey_type": "uinput",
        "do": "KEY_Y"
      },
      "MACRO_3": {
        "hotkey_type": "uinput",
        "do": "KEY_F16"
      },
      "MACRO_4": {
        "hotkey_type": "uinput",
        "do": "KEY_F17"
      },
      "MACRO_5": {
        "hotkey_type": "uinput",
        "do": "KEY_F18"
      },
      "MACRO_6": {
        "hotkey_type": "shortcut",
        "do": "alt+F1"
      },
      "MACRO_7": {
        "hotkey_type": "shortcut",
        "do": "alt+F2"
      },
      "MACRO_8": {
        "hotkey_type": "shortcut",
        "do": "alt+F3"
      },
      "MACRO_9": {
        "hotkey_type": "shortcut",
        "do": "alt+F12"
      }
    },
    "MEMORY_3": {
      "MACRO_1": {
        "hotkey_type": "uinput",
        "do": "KEY_F14"
      },
      "MACRO_2": {
        "hotkey_type": "uinput",
        "do": "KEY_F15"
      },
      "MACRO_3": {
        "hotkey_type": "uinput",
        "do": "KEY_F16"
      },
      "MACRO_4": {
        "hotkey_type": "uinput",
        "do": "KEY_F17"
      },
      "MACRO_5": {
        "hotkey_type": "uinput",
        "do": "KEY_F18"
      },
      "MACRO_6": {
        "hotkey_type": "shortcut",
        "do": "alt+F1"
      },
      "MACRO_7": {
        "hotkey_type": "shortcut",
        "do": "alt+F2"
      },
      "MACRO_8": {
        "hotkey_type": "shortcut",
        "do": "alt+F3"
      },
      "MACRO_9": {
        "hotkey_type": "shortcut",
        "do": "alt+F12"
      }
    }
  }
}
```

suabo, 2024-05-25:
I can verify the Problem and already found the Solution.
The Problem is a wrong uinput key code in data_mappers/supported_devices.py.

`KEY_PREVIOUS` needs to be `KEY_PREVIOUSSONG` and
`KEY_NEXT` needs to be `KEY_NEXTSONG`

I already tried it with a local rebuild and it works. I'll try to release a new version with the fix in a few days.

Thanks for reporting @Captn138

suabo, 2024-05-25:
I released version 0.4.2 with the fix.

Just download the new version and use the uninstall and install script to update and all media keys should work fine.

closed 2024-09-10 by suabo