JSubelj/g910-gkey-macro-support #88: Install on Linux Mint / Ubuntu

tsposato, 2024-08-11:
Has anyone got this working properly on Ubuntu?
I get into a python/dependency issue having to use a venv...
Also, I can't verify uinput module being loaded?

suabo, 2024-08-13:
I run ubuntu 22.04.4 LTS and it's working fine.
Can you please provide a more detailed error description?
How did you install it?

darkmatus, 2024-09-03:
If you running Ubuntu/Mint with latest LTS you need to use Python3.8 (eventually 3.9 works also) with venv.

First install python 3.8 from source if it is not possible to install it via apt.
`sudo apt-get update
sudo apt-get install wget build-essential checkinstall libreadline-gplv2-dev libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev libffi-dev zlib1g-dev
`
`
cd /usr/src
sudo wget https://www.python.org/ftp/python/3.8.16/Python-3.8.16.tgz
sudo tar xzf Python-3.8.16.tgz
cd Python-3.8.16
`
`sudo ./configure --enable-optimizations
sudo make altinstall
`

checking with:
`
python3.8 --version
`

Now clone the repository:
`
git clone https://github.com/JSubelj/g910-gkey-macro-support.git
cd g910-gkey-macro-support
`

create the venv:
`
# Navigate to the directory where you want to set up the project
cd /path/to/your/project

# Create a virtual environment using Python 3.8
python3.8 -m venv g910-env

# Activate the virtual environment
source g910-env/bin/activate
`

install it:
`
sudo your-venv-dir/bin/pip3.8 install --log ./install.log -e ./
`

Know modify the service file:
`
Environment="HOME=your-home"
ExecStart=/usr/bin/env your-venv-dir/bin/g910-gkeys
`

reload the deamon and restart the service:
`
sudo systemctl daemon-reload && sudo systemctl restart g910-gkeys.service
`

suabo, 2024-09-03:
Thank you for your detailed explanation @darkmatus 

I just checked my python3 version and I am using 3.10.12 as default version and also running the driver with it just using the provided install script. If you got the latest updates, ubuntu will use 3.12.3 which could not work with the driver, so the example to install it maybe could be the solution for you.

suabo, 2025-08-02:
After upgrading to the newest LTS, I can confirm that issue. You can't run it with > python3.8 and this is no longer supported with the ubuntu 24.04 LTS. The provided package on pypi can't be installed in global environment. So you are forced to use a virtual one.

But there is a more easy way to install python3.8 into your system by adding the deadsneaks ppa:
```
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt-get update
```
And installing with apt:
```
sudo apt install python3.8 python3.8-dev python3.8-venv
```

After installing python3.8 you have to download the repository or clone it with git:
```
git clone https://github.com/JSubelj/g910-gkey-macro-support.git g910-gkeys
```
Navigate into the folder of the repository `cd g910-gkeys` and create the virtual environment:
```
python3.8 -m venv venv
```
Activate it:
```
source venv/bin/activate
```
Now you are in that virtual environment, which is indicated by (venv) in the beginning of the command line.

Install dependecies into venv:
```
pip install python-uinput pyusb
```
Build the package in this venv:
```
python3 -m build
```
Install the package still in venv:
```
pip install dist/g910_gkeys-0.4.2.tar.gz
```
Now you have installed the package into an virtual environment.
To make sure the command got installed you can check for the file `venv/bin/g910-gkeys`

You can leave the virtual environment now:
```
deactivate
```
Now you have to modify the service file with you favorite editor at this path `/usr/lib/systemd/user/g910-gkeys.service`
Add the path to your virtual environment in the service file like so:
```
ExecStart=/usr/bin/env /path-to-repository/g910-gkeys/venv/bin/g910-gkeys
```
_Replace **path-to-repository** with the real path._

Last step will be reloading of user daemon and restarting the service:
```
systemctl --user daemon-reload && systemctl restart --user g910-gkeys.service
```
There is no need to run sudo other then adding the deadsneak ppa and installing python3.8!

One more time, thank you @darkmatus for providing this manual, since there was missing deactivating the venv and installing with root I just completed the manual and changed it to a user based installation like it is supposed to be. You should have executed the install script already to make sure config, service unit, uinput permissions and udev rules are in place. Otherwise it will probably not work.

Maybe you also can just install the provided pip package into your virtual environment instead of building the package from source. Either way you need to have python3.8 and a virtual environment to run the command in ubuntu 24.04.

If I have some more time I will make an update to the install script to run everything in a virtual environment by default to make it easier to install.